realthunder/FreeCAD_assembly3 #472: More Bugs Recomputing Sketches Made in Mainstream

zardozerr, 2021-02-20:
It does look like there are still some issues translating sketches that were originally made in mainstream .19.

I've isolated this sketch from a larger project. It's simply a side panel that needs to be padded. If you try to pad this sketch which seems to be valid, it will say that the wire isn't enclosed. If you open this up in mainstream .19, it will pad properly. I've attached the file. Overall, there isn't anything very complicated about this sketch that should cause issues.
[SketchTest.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6016023/SketchTest.FCStd.zip)

ceremcem, 2021-02-20:
https://user-images.githubusercontent.com/6639874/108605893-f9b4e480-73c7-11eb-9f3f-b03497a6e9c5.mp4

I think there was a missing line?

zardozerr, 2021-02-20:
@ceremcem 

Thanks for taking a look at this. Pretty weird, though, because if you open it up and don't do anything to it, do you still see a line there? I do, and it's even selectable as "edge 5".
[image]
[image]

Then try to pad it, and it doesn't work. If I add a line like you did "on top", then it works like it does in your video.

Can you try it without adding the line in mainstream .19? Because on that, it will pad without doing anything. It still seems to me to be some kind of bug in translating the sketch.

realthunder, 2021-02-21:
This is a tricky case. You have created a duplicated edge at the location shown below. You can use shortcut 'G, G' to bring up the geometry picker to show the coincident edges. The upstream FreeCAD uses a differently face making algorithm that somehow ignored this kind of error, which arguably isn't the right thing to do. In my branch, you can force using the old algorithm by setting `Pad.AllowMultiFace` to false.

https://user-images.githubusercontent.com/1207888/108612213-614b4e00-7421-11eb-8853-5acd4ad8960b.mp4

zardozerr, 2021-02-21:
@realthunder 
Interesting, thanks for the quick response. I had no idea that there was a geometry picker like that. And certainly because the pad worked in mainstream .19, I also had no idea there was an error and didn't bother to check.

ceremcem, 2021-02-21:
Is this issue resolved for you?

zardozerr, 2021-02-21:
Since it looks like it isn't really a bug, this is resolved, thanks.

closed 2021-02-21 by zardozerr