realthunder/FreeCAD_assembly3 #473: Compiling ended with error in project SketcherGui

Rabbit-sk, 2021-02-21:
Dear RealThunder, 

I tired to compile your branch on Windows, because i need to use fixed OBJ export #466, but after 2 hours and 5 minutes the compilation ended with error.

I followed instructions [here](https://github.com/realthunder/FreeCAD_assembly3/wiki/Build-Instruction), and [here](https://wiki.freecadweb.org/Compile_on_Windows). I was building with Visual Studio 16 (2019), and used these [FreeCADLibs_12.4.3_x64_VC17.7z](https://github.com/FreeCAD/FreeCAD/releases/download/0.19_pre/FreeCADLibs_12.4.3_x64_VC17.7z). The source code was cloned today from [here](https://github.com/realthunder/FreeCAD.git) after commit [b208947](https://github.com/realthunder/FreeCAD/commit/b20894732c490850cde482662b5fa4bd08c8c25f). 

[image]
[image]
[image]
[image]
[image]

Output:
`========== Build: 110 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========`

As you can see at the end of the Error List [Linker Tools Error LNK2019](https://docs.microsoft.com/en-us/cpp/error-messages/tool-errors/linker-tools-error-lnk2019?f1url=%3FappId%3DDev16IDEF1%26l%3DEN-US%26k%3Dk(LNK2019)%26rd%3Dtrue&view=msvc-160)  occured in project SketcherGui.

ReportView:
```
22:58:44  During initialization the error "No module named 'SketcherGui'" occurred in C:\Users\Jan\AppData\Roaming\FreeCAD\Mod\Part-o-magic\InitGui.py
22:58:44  Please look into the log file for further information
22:58:44  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "D:\SOFTWARE\FreeCAD\LinkBranch\Build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'SketcherGui'
22:58:44  No module named 'SketcherGui'
23:08:46  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "D:\SOFTWARE\FreeCAD\LinkBranch\Build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'SketcherGui'
23:08:46  Traceback (most recent call last):
  File "<string>", line 53, in Initialize
  File "D:\SOFTWARE\FreeCAD\LinkBranch\Build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)

23:08:46  No module named 'SketcherGui'
```

Also my build directory has size of 9,83 GB, while your daily builds have approximately 1,6 GB. Please advise me what to do to have build size comparable to yours. Thank you!

Regards

realthunder, 2021-02-22:
I just pushed a fix for this. Although it would be nice for you to build the binary yourself, but if you just want the OBJ fix, simply download [this](https://github.com/realthunder/FreeCAD/blob/LinkStage3/src/Mod/Arch/ArchCommands.py) file and copy it into `<Your_extracted_7z_directory>/Mod/Draft/`.

Rabbit-sk, 2021-02-22:
> simply download this file and copy it into `<Your_extracted_7z_directory>/Mod/Draft/`.

`ArchCommands.py` is not there, but I found it in `../Mod/Arch/`.

What an easy solution! I worked all night a few days ago and when I noticed that there was a fix, I didn't notice that it was a python file 🙈 and since then I have thought it needs to be built ... After a long wait for a new build , I decided to compile it myself, despite the fact that for the first time the process seemed complicated and lengthy. Although the solution in this case is much easier than compiling, it was worth learning new things and solving various problems. I will try the compilation again because I would like to have a debug build so I could provide more information to you in reports if needed.

Rabbit-sk, 2021-02-22:
Source code compiled successfully after commit [9deefac](https://github.com/realthunder/FreeCAD/commit/9deefac4f2748d4af44d5d856f605c92d736c8c3).

closed 2021-02-22 by Rabbit-sk