realthunder/FreeCAD_assembly3 #474: Dock window overlay, not resizing

snelweg, 2021-02-22:
When I am in overlay mode the treeview is always to narrow. I can not adjust it until I change with **F3** to non-overlay mode.

https://user-images.githubusercontent.com/1257901/108701129-d85b1200-7507-11eb-824f-2b8f1002954f.mp4

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2021-02-23:
It is not really the dockable widget size that clips the tree view. It is the tree view column size. There are two options regarding this behavior. If you disable resizable tree column, then the column will auto expand to show everything (but still limited by the dock widget size of course). If you prefer resizable column, you can turn off tree view header hiding.  Note that currently, the tree column size is not remembered. You'll need to change it on every start up. I'll add column size persisting later.

[image]

snelweg, 2021-02-23:
@rt Wow, so many options. I didn't think this was adjustable by the user. But you made it configurable, as always you have the user in mind. Thanks for that.
I adjusted the checkboxes and now it works as expected.

thanks!

closed 2021-02-23 by snelweg