realthunder/FreeCAD_assembly3 #475: Issues with external monitors (scaling / menu placement)

oliof, 2021-02-22:
I recently added a 34" screen to my workplace so I have more screen real estate for FreeCAD than the 1920x1080 on my external laptop screen. Now I have 

Laptop screen: 1920x1080
External screen: 3440x1440

When I open FreeCADlink.exe (Daily 2021.02.05 release) on the external screen, the UI font size is really tiny and fuzzy, unlike other programs, but just readable. When I disconnect the external screen, all windows go back onto the laptop screen and get rescaled to look about right there. FreeCADlink then looks about the same as if I had started it on the Laptop screen.

When I then reconnect the external screen and move FreeCADlink.exe back to the external screen, all UI items (icons, fonts etc.) are very big. I.e., they are not scaled to approximate the visual size they had on the internal screen.

But there is more: When I click on a menu entry, the menu bar is drawn on the laptop screen, approximately the location the FreeCADlink window was before I dragged it over to the external monitor. I can then mouse over to the laptop screen and use the menu, but obviously this is not a good way to use a program.

luzpaz, 2021-02-22:
Please add your full [About](https://wiki.freecadweb.org/About) info

oliof, 2021-02-22:
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git)
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: German/Germany (de_DE)

realthunder, 2021-02-23:
I just tried with an external monitor with my laptop. It does seem to misbehave when move across different monitors. But I don't have a 4K monitor, so can't really duplicate what you have described. Does this also happen with upstream FreeCAD?

Are you using the extended display mode, meaning that two monitor display different content? Does FreeCAD behave correctly, if you set your external monitor as the primary display?

ceremcem, 2021-02-23:
> But I don't have a 4K monitor

Ready to add `waiting cargo` label.

oliof, 2021-02-23:
- The current 0.19 prerelease version of FreeCAD exhibits the same issue(s):

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24219 (Git)
Build type: Release
Branch: master
Hash: 8c26baebab320b8c1c3279bc8eb34a1eb6c7a363
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

- I am using extended mode
- When setting the external screen as primary, I am not getting the menu misplacement issue, even when moving FreeCADlink to the lap top screen, but the UI is still changing size drastically.

oliof, 2021-02-23:
OK, I was able to work around the scaling issues by enabling the following Application Properties:

Properties" > _/Compatibility\_ > --Settings-- > [Change high DPI settings] opens a secondary dialog with the following controls:

Program DPI
[X] Use this setting ... I open this program

High DPI scaling override
[X] Override high DPI... [System (Enhanced)]

The remaining issue is the menus popping up on the internal screen when it's set to be the primary screen if the external screen is disconnected and reconnected. For that, the workaround of setting the external screen to be primary works (in fact, toggling primary back and forth fixes where the menus are drawn).

akamensky, 2021-02-25:
This is generally issue with FreeCAD UI, not specific to this branch.

It is not about the resolution per-se, it is about HiDPI (or what's called Retina on macos), where you have double density of pixels.

There were a bunch of fixes in upstream for this, but the support for HiDPI is still patchy at best (some places look ok, others are not).

clwe, 2021-06-23:
I have a similar issue on linux: 

with small menu items looks almost ok: 
[image]

with medium menu icon size looks strange: 
[image]

in both cases I think the start page and the menu items are rescaled to be too big.
my external screen is the primary screen and it's resolution is 1920x1080. 
The laptop screen is also 1920x1080, but when I drag freecad there it is rescaled to have very tiny fonts:

[image]

here, the menu items and start page are looking good though.

OS: Ubuntu 21.04 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)