realthunder/FreeCAD_assembly3 #477: Sketch creation error with 2012.02.24

sigxcpu76, 2021-02-24:
File>New
Part Design
Create Body
Create Sketch

error in report:
```
23:12:15  <PartDesign> CommandBody.cpp(148): Ignore selection with no solid shape: Unnamed#Body.
```

Creating a body-less sketch (which is not "the way" since FreeCAD 17, I think) is allowed though.

[image]

realthunder, 2021-02-25:
There is some mis-interpretation of selection when creating the sketch. As a temporary work around, simply clear the selection before creating the sketch.

sigxcpu76, 2021-03-06:
Is the new way to attach initial sketch intentional after this fix?

Previously, creating a sketch would bring this mode (select feature).
[image]

After this fix it starts in this mode (select attachment).

[image]

realthunder, 2021-03-06:
Yes, it is intentional. The new behavior creates the sketch first, and opens up its attachment editing task panel, which allows you to select not only origin or plane, but also datum. In fact, you can select any object, even those outside of the body, and it will be auto imported into the body.

sigxcpu76, 2021-03-06:
Got it. It is a bit harder to attach to standard planes. Now you need to open Origin, select a plane, then click "New Sketch". As far as I've seen in various YT videos, that is the workflow with the first sketch in a body. Subsequent sketches can/may have more complex attachment modes.

oficinerobotica, 2021-06-07:
Tested with:
```OS: KDE neon User Edition 5.21 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```
and the issue isn't present anymore.

closed 2021-06-07 by oficinerobotica