realthunder/FreeCAD_assembly3 #479: Issue to click buttons with stylesheets

bf5man, 2021-02-27:
I noticed that if you apply a stylesheet to Freecad, the interface buttons will only be clickable if you click their text, but clicking in the button area beside the text won't be registered as a click.

**Steps To Reproduce:**

1. Load the Preferences->General->Style Sheet->Set it to _Behave Blue_
2. Click the _Apply_ button (the new style is applied)
3. Try to click the _OK_ button, by clicking in the button area but beside the "ok" text
4. Notice that the click won't be registered
5. Try to click the _OK_ button again, but this time click the "ok" text
6. Notice that the click will be registered.

Thanks!

OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0224.23964 +3289 (Git)
Build type: Release
Branch: LinkStage3
Hash: 2dd87c255154b308a7681c1a7c5740151f340b0a
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: French/Canada (fr_CA)

realthunder, 2021-02-27:
It is a Qt 5.15.1 [bug](https://bugreports.qt.io/browse/QTBUG-88276). And looks like Qt 5.15.2 [fixed](https://bugreports.qt.io/browse/QTBUG-86587) it. This seems to be a very annoying problem, and it will affect upstream build as well. @apeltauer is it possible for you to release a libpack with Qt 5.15.2 or later maybe?

luzpaz, 2021-02-27:
@looooo & @sgrogan just giving a heads up about this

StudioPetrikas, 2021-03-03:
As a (very) temporary solution, I suggest editing the stylesheet, finding (line ~1400)

`QPushButton {`

and adding a line

`font-weight: 900;`

It makes the push-button text Heavy, so you have a much (much) higher chance of hitting it. Been working with the broken button for a full day and it's been driving me nuts.

realthunder, 2021-03-09:
I'll downgrade to Qt 5.12 for windows in the next release.

HaasNL2, 2021-03-09:
I also edited my stylesheet for a workaround for now. Removed the padding and made the text thick.
It needs to be copied into the "data\Gui\Stylesheets" dir and then selected in the settings.

[ProDark - button workaround.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6108976/ProDark.-.button.workaround.zip)

oficinerobotica, 2021-06-09:
Tested with: 
```
OS: KDE neon User Edition 5.21 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```
and the issue isn't present anymore.

closed 2021-06-09 by oficinerobotica