realthunder/FreeCAD_assembly3 #482: use the expression editor for every input

snelweg, 2021-02-27:
I never knew consciously that this is a real pain. Now I see somebody else observing the same issue I kindly draw your attention to this. Would be something that would really suit the LinkStage branch.

> "Not really a "bug", but I have never understood why there is a separate expression editor compared to the usual constraint input box. I really don't think it would be difficult to use the expression editor for every input to save so many clicks when doing parametric modeling. There are actually quite a few UI quirks like this (the Add/Remove button when creating chamfers is so goofy!) that would help usability so much and seem like low hanging fruit to me."

From a redit thread: https://www.reddit.com/r/FreeCAD/comments/lsv0qe/on_what_freecad_bug_would_you_set_out_a_bounty/gotyqhr?utm_source=share&utm_medium=web2x&context=3

Rabbit-sk, 2021-02-28:
Hi @snelweg, 

Would it be enough for you to turn on `In place editing` and possibly set your own shortcut to invoke the expression editor? By default it is `=` key. I personally prefer `F` key.
[image]
[image]
You can also check out the discussion here #397.

snelweg, 2021-02-28:
It could be more then enough. Still think it is a valid request to have the  expression editor to be used for all the inputs.
I already use the '=' shortkey, my point is that every input could use the expression editor.

Rabbit-sk, 2021-02-28:
> Still think it is a valid request to have the expression editor to be used for all the inputs.

So why close this issue? Let's discuss.

You didn't specify the desired look of the input field, so I'm just thinking. I don't know how is this solved in another 3D CAD software, so I will consider the look and functionality of Expression Editor as it is now in FreeCAD.

Expression editor consists of more GUI components than just a parameter field:
[image]
It can be aslo resized if longer expression is going to be edited:
[image]

Consider things mentioned above and these input fields: 
[image]
I think that it would take up much more space.

Or every input field, after clicking to it, could look like this:
[image]
but it would be impractical if someone want to enter just a constant number and then want to enter some numer to the next field. 

Recently I do not need expressions a lot, and simple input field is practicall. Some time ago I was using expresions a lot, and for me invoking Expression Editor was also impractical. For me personally, it was enough that it is possible to choose my own shortcut to invoke Expression Editor. As I see it, someone else would welcome it if this step were not needed. The needs and preferences are various from person to person and sometimes also from project to project... So for everyone to be happy, I suggest this little change:

In `Preferences` next to `Editor trigger shortcut` field add checkbox with text: "Input field activation". If chcecked, shortcut field would be grayed and every by default activated or clicked input field would look like this:
[image]

Would you welcome such a solution?

snelweg, 2021-02-28:
Yes looks great and I am sure it will function very smoothly.
You are right that if every input was to be changed to a standard expression editor there would be problems with the screen estate.
(and I closed it because it was a suggestion from me, didn't look like it would get many supporters)

Rabbit-sk, 2021-02-28:
Now we have the specification of desired change. I wonder what @RealThunder thinks about it. I think that you could reopen this isssue.

snelweg, 2021-02-28:
> Now we have the specification of desired change. I wonder what @realthunder thinks about it.

Dear RealThunder, please kindly consider this request. I know I would be happy to have such an enhancement and I am sure others would be helped with this too.

Rabbit-sk, 2021-02-28:
> Yes looks great and I am sure it will function very smoothly.

I realized that this proposal should be adjusted a bit. Expression Editor overlays several dialog items, and it would not be appropriate to overlap them immediately after opening the feature dialog. Input fields in feature dialogs, which are initially activated should be excluded from this behavior, and in place Expression Editor should appear later, based on user interaction, for example when the user starts typing a value.

snelweg, 2021-02-28:
> I realized that this proposal should be adjusted a bit. Expression Editor overlays several dialog items, and it would not be appropriate to overlap them immediately after opening the feature dialog. 

Yes, that's why I said:
"You are right that if every input was to be changed to a standard expression editor there would be problems with the screen estate."

Will wait for RT's reaction. Thanks for your input and making the idea concrete.

realthunder, 2021-02-28:
The in place expression editor is having some trouble on some system (See [here](https://forum.freecadweb.org/viewtopic.php?p=481977#p481977)). So I have just disable it by default in recent release. User can still enable it with preference.

I don't think auto invoking expression editor is practical. What I can try is for the spin box to be smarter and check for possible expression entering. I'll experiment on that in the future.

snelweg, 2021-03-01:
> I don't think auto invoking expression editor is practical. What I can try is for the spin box to be smarter and check for possible expression entering. I'll experiment on that in the future.

Ok, thanks. It was just an idea, not all of my ideas are good or practical. Take your time on this one, no hurry.

closed 2021-03-01 by snelweg