realthunder/FreeCAD_assembly3 #483: Poliline in sketch "M" cycle not work.

dgrabulosa, 2021-02-27:
As expectet push m key not cycle polyline, but shift+m works

snelweg, 2021-03-04:
I found that it sometimes does work. What version ar eyou using?

wohltat, 2021-03-06:
i see the same behavior with the latest release

OS: Linux Mint 19 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0301.23964 +3306 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 32f78691fae500e494879493abc2b74cff9585d3
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United Kingdom (en_GB)

dgrabulosa, 2021-05-01:
It works as expected, thankyou.

closed 2021-05-01 by dgrabulosa

wohltat, 2021-07-31:
Still doesn't work on my System. Neither M nor shift + M. I don't why.
Do you know something that i can do about it?

OS: Linux Mint 20.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

realthunder, 2021-07-31:
Probably some conflict shortcut setting. Can you please post your `~/.FreeCAD/link.user.cfg` file?

wohltat, 2021-07-31:
Thanks for the hint. I had a look into it and recognized that my Shortcut for Polyline is "P".

So there really was no problem in the first place. I can switch the modes with "P" key just fine.