realthunder/FreeCAD_assembly3 #485: "Fit all" does not work on some arrays

Alomphega, 2021-03-01:
Hello,

As asked on forum by RealThunder, I post the bug here : 
the "Fits the whole content on the screen (V, F)" button does not work at all. The same with "Fit all" menu, 

BUT ONLY on newly created files with one or more arrays under a visible Boolean, and these arrays visibility toggled OFF (when the Boolean is visible, it does not fit if we hit "Fit all") : 
[image]

Hoping this will help a bit,
Guy

realthunder, 2021-03-01:
Can you please provide a file showing the problem. BTW, how do you manage to create a boolean with features outside of the body?

Alomphega, 2021-03-02:
Hello,

Please find the test file made for you in attachment.
To answer you : I did not manage to create a boolean with features outside of the body, I just do all in bodies, as you will able to see this in that test file. Please feel free to ask your questions further if any.
[20210302_test_for_RealThunder_LB2210227.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6067119/20210302_test_for_RealThunder_LB2210227.zip)

Cordially,
Guy

realthunder, 2021-03-02:
I have fixed view fit problem.

On an unrelated issue, though, the Part Mirror and Draft array are not PartDesign features. So by right, they should not be allowed inside a body. Can you recall your steps of adding the mirror to the boolean feature? Are you adding those features using my branch?

Alomphega, 2021-03-04:
Hello,
Yes, I've done this test file with your 20210227 Link Branch test version... Just did the sketch, then the pad, then the array, then the mirror.
Hope this will help you to understand something in what you call an issue for this.
Cordially,
Guy