realthunder/FreeCAD_assembly3 #486: Not enough memory available (on save)

theory14, 2021-03-01:
With both 2021.03.01 and 2021.02.27-2 I get frequent errors "Not enough memory available" when attempting to save.  This did not occur with 2021.02.05 or with 0.18 or 0.19-pre from freecadweb.  Autosave does seem to be saving the file.  The error seems to occur randomly, but frequently. 

Manual save attempt with error generates the following:
13:07:03  <Exception> Not enough memory available

I see this also in the output:
13:07:49  Save AutoRecovery file: Needed time: 0ms

This occurs with trivial files -- single simple sketch and single extrusion.

OS:  macOS 11.2.2
RAM:  32 GB

realthunder, 2021-03-01:
02.27-2 is released only for Windows. Do you mean you've started to experience the problem since 02.27 release on MacOS? Is this problem happening on every file save or just some? Any particular path or file name you are using? Can you please try the latest upstream FreeCAD and see if there also this problem?

theory14, 2021-03-02:
- It was 02.27, not -2.  I grabbed it on the day when -2 was released and didn't pay close enough attention that -2 was just windows and the Mac release was just the base 02.27.
- This error occurs randomly, but frequently.  
- This does not occur with 2021.02.05 or with 0.18 or 0.19-pre (rev 24267) from upstream
- This happens with files attempted to save to different paths
- Sometimes when this problem appears, editing and closing an existing sketch  (making no changes) allows a save to succeed.
- I also just noticed that a file is actually saved when this error pops up, but it is named oddly.  Four (4) attempts to save that generated the error message also created these files -- one for each "failed" save.
birdfeederbase.FCStd.a11d3f08-5fb1-4f25-b936-4b8238afbff7
birdfeederbase.FCStd.c1c1b442-845c-40e2-bd91-353c0cbbef59
birdfeederbase.FCStd.d0eec67e-94b6-4d80-ad2c-47540266687e
birdfeederbase.FCStd.e2d86ea9-8d96-4aa6-9795-2071fac77aea
- I looked at the output from dtruss on the FreeCAD process, when the save error occurred and didn't see anything of use.  If there is any other debugging I can do, please feel free to ask.

Also, thanks for all of your work on the project.

StudioPetrikas, 2021-03-03:
This happens on the Windows release as well.
Currently running:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0301.23964 +3306 (Git)
Build type: Release
Branch: LinkStage3
Hash: 32f78691fae500e494879493abc2b74cff9585d3
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)

"Not enough memory available" and the viewport background turns white. Clicking save a couple more times or entering/exiting sketch helps to by pass. It wouldn't be too much of a problem if it wasn't for the white background 'flashbang'. Closing and reloading the file fixes the problem for some time.

HaasNL2, 2021-03-03:
Same issue have it at save!
[image]

ceremcem, 2021-03-03:
@rabbit-sk Do you experience the same issue with your latest build?

Rabbit-sk, 2021-03-03:
Hi @ceremcem, 

Now I'm not using the newest version for creating  models (except  the fixed exporting). I will switch to some new version when I will have more time to test it. But to check this issue I will use ofcourse the latest daily build:

[image]

Saved without problem. Nothing appeared in the Report View.

OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0301.23964 +3306 (Git)
Build type: Release
Branch: LinkStage3
Hash: 32f78691fae500e494879493abc2b74cff9585d3
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Slovak/Slovakia (sk_SK)

HaasNL2, 2021-03-04:
It seems to be fixed when starting with administration rights.

realthunder, 2021-03-04:
There is probably some kind of memory leak somewhere. The upstream has quite a few fixes recently about memory problem. I'll make the next release with another merge with the upstream.

theory14, 2021-03-07:
I just tried 0307 and the issue is still present.  If there is any data I can provide to help troubleshoot, please let me know.

realthunder, 2021-03-08:
I have finally identified the problem (I think), and hopefully fixed it. I'll make a new release late today.

Meanwhile, to work around this problem, make sure you do not have any selection or pre-selection (i.e. move your mouse away from the 3D view) before saving the document. Most of your crash happens when saving the file while your mouse hovering above some geometry.

closed 2021-03-09 by theory14