realthunder/FreeCAD_assembly3 #487: Feature Request: Speeding up Aliasing creation with "Imply Alias"

deftdawg, 2021-03-03:
It would speed aliasing if the Alias command or a varant of the Alias command implied default values from other cells in the spreadsheet.

If the preceeding column value is not an Alias and not a number, that should be taken as the default `Alias of this cell` value.  
If the cell in the preceeding row in the same column is an Alias with a unit, that should be taken by default as `Display unit`.

So for example:

|Row | A        | B           |  C |
|:-------------:|:-------------:| :------------:| ------------|
| 1 | width | 25 mm |  `imply "width" as Alias value; "mm" would have to be entered as first Display unit in the column` | 
| 2 | height |  50 mm | `imply "height" as Alias value; imply "mm" as Display unit from B2`  |

Not sure the best way to make this behiaviour accessible, maybe an alternate click on the cell to be aliased with `Imply Alias`  when the values to guess are available would be a good place to start. 

It might also be good to show the Alias in a different colour if the cell in the proceeding column no longer matches the value of the alias to help avoid confusion; or maybe if the column value and the Alias name could be linked somehow.

snelweg, 2021-03-04:
Finaly! I always thought I was the only one struggling with naming my parameters. The macros that are available are not very good. It would be great if RT would shed some magic onto this.

ceremcem, 2021-03-04:
> If the preceeding column value is not an Alias and not a number, that should be taken as the default Alias of this cell value

That's a conflicting behavior with current UX, IMO. Yes, I also want what you want, but that may not be that easy.

My proposal is as follows: 

Typing `=` after a string will make the cursor switch to next column and will set that cell's alias as the previous cell's content. 

Example: 

To create above table, do these: 

1. Click A1
2. Type `width=` + `Enter`. This will erase last `=` character and make A1 content `width`, proceed to B1
3. Type `25mm` + `Enter`. Previous action will set this cell's alias to `width`. Since `Enter` would normally proceed to next row, cursor will be end up in A2 after step `3`. 
5. Type `height=` + `Enter`
6. Type `50mm` + `Enter`

deftdawg, 2021-03-04:
> That's a conflicting behavior with current UX, IMO. 

Being I just started with FreeCAD 2 days ago, I'm a bit confused why that would be a conflicting UX behaviour...  Can you elaborate a bit?

> 
> My proposal is as follows:
> 
> Typing `=` after a string will make the cursor switch to next column and will set that cell's alias as the previous cell's content.
 
> Example:
> 
> To create above table, do these:
> 
> 1. Click A1
> 2. Type `width=` + `Enter`. This will erase last `=` character and make A1 content `width`, proceed to B1
> 3. Type `25mm` + `Enter`. Previous action will set this cell's alias to `width`. Since `Enter` would normally proceed to next row, cursor will be end up in A2 after step `3`.
> 4. Type `height=` + `Enter`
> 5. Type `50mm` + `Enter`

This seems less initutive to me, I think from a noobies prespective there should be some visual cue to activate the behaviour like a alt-mouse menu click + select `imply alias` or menu entry or a toolbar button.    How would a newbie such as myself discover that placing a magic `=` at the end of a string would trigger that behaviour without reading a ton of docs?

realthunder, 2021-03-04:
I am not sure if all of you knows that there is the 'Edit mode' option for each cell in my branch. Try that and you'll find it much easier to edit fields.

For this auto alias feature, I would add two new edit modes, `Auto alias', and `Auto alias vertical`. Once any cell is in this edit mode, say `Auto alias`, then it will assign its left sibling cell an alias using the cell content. Furthermore, it will activate the `Auto alias` mode when adding a new cell to its bottom. You can end the chain by manually turning off `Auto alias` mode in any cell. `Auto alias vertical` behavior similarly in a different orientation.