realthunder/FreeCAD_assembly3 #489: Segfaults during editing

tomster, 2021-03-04:
I'm running ``./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210301-glibc2.12-x86_64.AppImage -l`` and after some currently not precisely reproducible operations, the application segfaults.

this is running on a ubuntu 20.04. system. Is there any other information that would be helpful?

```Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f949b84b210]
#1  0x7f949bd41c4e in QMetaObject::methodCount() const from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x4
#2  0x7f949a9838c6 in PySide::getMetaDataFromQObject(QObject*, _object*, _object*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib64/../lib/libpyside2.cpython-38-x86_64-linux-gnu.so.5.13+0x126
#3  /tmp/.mount_FreeCA00TJZl/usr/lib/python3.8/site-packages/PySide2/QtWidgets.cpython-38-x86_64-linux-gnu.so(+0x4679e8) [0x7f94860dd9e8]
#4  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(_PyObject_GetMethod+0x14c) [0x7f949d2e99fc]
#5  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x7ba) [0x7f949d334c6a]
#6  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(_PyFunction_Vectorcall+0xfb) [0x7f949d3015bb]
#7  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(+0x17ad13) [0x7f949d31dd13]
#8  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(PyVectorcall_Call+0x71) [0x7f949d2d51c1]
#9  /tmp/.mount_FreeCA00TJZl/usr/lib/python3.8/site-packages/PySide2/QtWidgets.cpython-38-x86_64-linux-gnu.so(+0x6738da) [0x7f94862e98da]
#10  0x7f949bd3d019 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0xcd
#11  0x7f949c5657e1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Widgets.so.5+0xcb
#12  0x7f949c56b31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#13  0x7f949da20399 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib64/libFreeCADGui.so+0x89
#14  0x7f949bd3f606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x86
#15  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Gui.so.5(+0x10c105) [0x7f949c0f2105]
#16  0x7f949c0f21ad in QInputMethod::cursorRectangle() const from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Gui.so.5+0x17
#17  /tmp/.mount_FreeCA00TJZl/usr/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so(+0x8db6) [0x7f948f6acdb6]
#18  0x7f949bd55a18 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x6c0
#19  0x7f949c0f2351 in QInputMethod::update(QFlags<Qt::InputMethodQuery>) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Gui.so.5+0x55
#20  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Widgets.so.5(+0x19e774) [0x7f949c58f774]
#21  0x7f949bd55a18 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x6c0
#22  0x7f949c63523e in QWidgetLineControl::cursorPositionChanged(int, int) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Widgets.so.5+0x3c
#23  0x7f949c636105 in QWidgetLineControl::emitCursorPositionChanged() from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Widgets.so.5+0x1b
#24  0x7f949c63625d in QWidgetLineControl::setSelection(int, int) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Widgets.so.5+0xfd
#25  0x7f949c63249e in QLineEdit::setSelection(int, int) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Widgets.so.5+0x64
#26  /tmp/.mount_FreeCA00TJZl/usr/lib/python3.8/site-packages/PySide2/QtWidgets.cpython-38-x86_64-linux-gnu.so(+0x40f832) [0x7f9486085832]
#27  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(PyCFunction_Call+0xf5) [0x7f949d327725]
#28  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x230) [0x7f949d2cf550]
#29  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x4ddd) [0x7f949d33928d]
#30  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x2ea) [0x7f949d2d0d9a]
#31  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(+0x17ab4d) [0x7f949d31db4d]
#32  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x392) [0x7f949d334842]
#33  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(_PyFunction_Vectorcall+0xfb) [0x7f949d3015bb]
#34  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libpython3.8.so.1.0(PyVectorcall_Call+0x71) [0x7f949d2d51c1]
#35  0x7f949a963bad in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib64/../lib/libpyside2.cpython-38-x86_64-linux-gnu.so.5.13+0x9d
#36  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib64/../lib/libpyside2.cpython-38-x86_64-linux-gnu.so.5.13(+0x1a081) [0x7f949a96d081]
#37  0x7f949bd55a7f in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x727
#38  0x7f949bd5c912 in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x2a
#39  /tmp/.mount_FreeCA00TJZl/usr/lib/python3.8/site-packages/PySide2/QtCore.cpython-38-x86_64-linux-gnu.so(+0x2ad94d) [0x7f948c73794d]
#40  0x7f949bd54533 in QObject::event(QEvent*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x5b
#41  /tmp/.mount_FreeCA00TJZl/usr/lib/python3.8/site-packages/PySide2/QtCore.cpython-38-x86_64-linux-gnu.so(+0x2ad4d5) [0x7f948c7374d5]
#42  0x7f949c5657f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#43  0x7f949c56b31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#44  0x7f949da20399 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib64/libFreeCADGui.so+0x89
#45  0x7f949bd3f606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x86
#46  0x7f949bd6d8a4 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x24c
#47  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5(+0x1dbf54) [0x7f949bd6df54]
#48  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f949813198d]
#49  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/./libglib-2.0.so.0(+0x54c21) [0x7f9498131c21]
#50  /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f9498131cc1]
#51  0x7f949bd6e0e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x5e
#52  0x7f949bd3c019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0x187
#53  0x7f949bd3fe8f in QCoreApplication::exec() from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib/libQt5Core.so.5+0xfd
#54  0x7f949d99b53e in Gui::Application::runApplication() from /tmp/.mount_FreeCA00TJZl/usr/bin/../lib64/libFreeCADGui.so+0x40e
#55  /tmp/.mount_FreeCA00TJZl/usr/bin/FreeCADLink(+0x44bf) [0x5638d52704bf]
#56  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f949b82c0b3]
#57  /tmp/.mount_FreeCA00TJZl/usr/bin/FreeCADLink(+0x47d9) [0x5638d52707d9]```

ceremcem, 2021-03-04:
Please add the relevant file and add the "reproduction" steps even though those steps are not always reproducing the crash.

realthunder, 2021-03-04:
Yes a demo file will be helpful. Also, the crash happen inside PySide, i.e. the Python binding of Qt. Can you please describe what you are doing using what workbench at the time of crash. Judging from the call stack, it seems that you are using mouse to select some text from some line edit widget?

tomster, 2021-03-04:
yes, I'm following the skeleton tutorial (https://github.com/realthunder/FreeCAD_assembly3/wiki/Create-Skeleton-Sketch) for a project that i think that approach would be useful for. of course, now it's not crashing :) but i have everything ready, for when it does.

tomster, 2021-03-04:
oh and since it's the first time i'm interacting here, let me take the opportunity to thank you @realthunder for your awesome work and all your contributions. i'm really, really happy with FreeCAD in general and your branch with asm3 in particular!

realthunder, 2021-03-04:
Thanks for the appreciation!

So maybe the crash happens when you are editing some geometry in Draft workbench. Can you recall some more details?

BTW, if you master sketch is planar, it will probably be easier to just create a Sketch using Sketcher workbench. Create a circle for each attachment point in the sketch, and then constraint parts to the sketch using `Attachment` constraint. Draft geometry is more 3D friendly, although the tutorial is in 2D.

https://user-images.githubusercontent.com/1207888/110036088-72785100-7d77-11eb-9243-63408a47921c.mp4

tomster, 2021-03-04:
wow, that's amazing. i did not know that is possible. this is indeed a better approach for my particular problem. i will try it out asap. no further crashes during the past hour btw :)

tomster, 2021-03-05:
so, i started from scratch with a simpler model. and before getting to the part of creating and manipulating a skeleton sketch as in the video above i experienced another segfault.
this time i have uploaded the model. i'm not entirely sure, what i was doing exactly at the time of the crash. i had just finished fixing the ``PlaneAlignment005`` and ``PlaneAlignment006`` (replacing referenced elements). i might have just saved it (as is now my habit) with ctrl-s and/or i was re-orienting the view to continue with the next task.

```57.169924 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint001 (PlaneAlignment001) between Unnamed1#Assembly002 (foot001) and Unnamed1#Assembly001 (leg001), 2
0.000671 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint002 (PlaneAlignment002) between Unnamed1#Assembly001 (leg001) and Unnamed1#Assembly002 (foot001), 3
0.00137 <asm3.sys> sys_slvs.py(68): dof remaining: 0
49.150075 <asm3.sys> sys_slvs.py(68): dof remaining: 6
35.899921 <asm3.sys> sys_slvs.py(68): dof remaining: 6
5.729498 <asm3.sys> sys_slvs.py(68): dof remaining: 6
7.974241 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint004 (PlaneAlignment004) between Unnamed1#Link_i1 and Unnamed1#Assembly003 (foot_connector), 2
0.001986 <asm3.sys> sys_slvs.py(68): dof remaining: 4
24.258411 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint001 (PlaneAlignment001) between Unnamed1#Assembly002 (foot001) and Unnamed1#Assembly001 (leg001), 2
0.000707 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint002 (PlaneAlignment002) between Unnamed1#Assembly001 (leg001) and Unnamed1#Assembly002 (foot001), 3
0.001323 <asm3.sys> sys_slvs.py(68): dof remaining: 0
0.053057 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint004 (PlaneAlignment004) between Unnamed1#Link_i1 and Unnamed1#Assembly003 (foot_connector), 2
0.001502 <asm3.sys> sys_slvs.py(68): dof remaining: 4
3.103366 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint004 (PlaneAlignment004) between Unnamed1#Link_i1 and Unnamed1#Assembly003 (foot_connector), 2
0.001465 <asm3.sys> sys_slvs.py(68): dof remaining: 4
2.816893 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint004 (PlaneAlignment004) between Unnamed1#Link_i1 and Unnamed1#Assembly003 (foot_connector), 2
0.001425 <asm3.sys> sys_slvs.py(68): dof remaining: 4
26.339401 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint004 (PlaneAlignment004) between Unnamed1#Link_i1 and Unnamed1#Assembly003 (foot_connector), 2
0.000675 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i0, 2
0.000342 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i1, 3
0.002258 <asm3.sys> sys_slvs.py(68): dof remaining: 1
28.14898 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint001 (PlaneAlignment001) between Unnamed1#Assembly002 (foot001) and Unnamed1#Assembly001 (leg001), 2
0.00076 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint002 (PlaneAlignment002) between Unnamed1#Assembly001 (leg001) and Unnamed1#Assembly002 (foot001), 3
0.001373 <asm3.sys> sys_slvs.py(68): dof remaining: 0
0.070783 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint004 (PlaneAlignment004) between Unnamed1#Link_i1 and Unnamed1#Assembly003 (foot_connector), 2
0.000722 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i0, 2
0.000352 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i1, 3
0.002195 <asm3.sys> sys_slvs.py(68): dof remaining: 1
4.451491 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint004 (PlaneAlignment004) between Unnamed1#Link_i1 and Unnamed1#Assembly003 (foot_connector), 2
0.000672 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i0, 2
0.00033 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i1, 3
0.002314 <asm3.sys> sys_slvs.py(68): dof remaining: 1
5.685002 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint004 (PlaneAlignment004) between Unnamed1#Link_i1 and Unnamed1#Assembly003 (foot_connector), 2
0.000658 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i0, 2
0.000339 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i1, 3
0.000867 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint006 (PlaneAlignment006) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i0, 3
0.003794 <asm3.sys> sys_slvs.py(68): dof remaining: 0
125.509321 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint004 (PlaneAlignment004) between Unnamed1#Link_i1 and Unnamed1#Assembly003 (foot_connector), 2
0.000675 <asm3.sys> system.py(357): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i0, 2
0.000339 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i1, 3
[... cut ...]
0.000338 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint005 (PlaneAlignment005) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i1, 3
0.000621 <asm3.sys> system.py(213): auto relax Unnamed1#Constraint006 (PlaneAlignment006) between Unnamed1#Assembly003 (foot_connector) and Unnamed1#Link_i0, 3
0.002637 <asm3.sys> sys_slvs.py(68): dof remaining: 0
realloc(): invalid next size
/tmp/.mount_FreeCAPmvaCf/AppRun: line 43: 3915897 Aborted                 (core dumped) "$PREFIX/bin/$name" "$@"
```
[wardrobe.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6090916/wardrobe.zip)

tomster, 2021-03-05:
btw i realise that this new segfault is probably technically not related to the first one, but when i was about to open a new issue i realized, the title of that would have been the same, so maybe we can collect any such segfaults here for the time being and you can sort it out :)

tomster, 2021-03-07:
i just tried out the new ``20210307`` release and immediately got a another segfault. this time during saving the document after performing a "transform" operation.

the last output before the crash was ```
0.000119 <asm3.sys> sys_slvs.py(68): dof remaining: 0
malloc(): invalid size (unsorted)
/tmp/.mount_FreeCAGWmtFF/AppRun: line 43: 75794 Aborted                 (core dumped) "$PREFIX/bin/$name" "$@"
```
i then restarted the application, recovered the file and performed the same operation again and saved - but this time it did *not* crash, so it's still unclear how to reproduce the crash.

[bench.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6097453/bench.zip)

tomster, 2021-03-07:
i then wanted to start the design from scratch in an empty document. created a new document, saved it. then copied the spreadsheet from the original document (drag and drop with ctrl). renamed the spreadsheet to ``data`` and saved and then it crashed like so:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f08224e6210]
#1  0x7f08220a4257 in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/../lib/libCoin.so.80+0x137
#2  0x7f0821e7a149 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#3  0x7f0821e81d96 in SoGLRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/../lib/libCoin.so.80+0x186
#4  0x7f0824981014 in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x64
#5  0x7f0821e7aac2 in SoAction::apply(SoPathList const&, int) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/../lib/libCoin.so.80+0x5b2
#6  0x7f082497e7df in Gui::SoBoxSelectionRenderActionP::apply(Gui::SoBoxSelectionRenderAction*, SoPathList const&, int) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x10f
#7  0x7f082497e859 in Gui::SoBoxSelectionRenderAction::apply(SoPathList const&, int) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x29
#8  0x7f0821e815d9 in SoGLRenderActionP::renderSingle(SoNode*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/../lib/libCoin.so.80+0x279
#9  0x7f0821e81aae in SoGLRenderActionP::render(SoNode*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/../lib/libCoin.so.80+0x11e
#10  0x7f0821e81cb2 in SoGLRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/../lib/libCoin.so.80+0xa2
#11  0x7f0824981014 in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x64
#12  0x7f0821e7ae2a in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/../lib/libCoin.so.80+0x2ea
#13  0x7f082497ebca in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x2a
#14  0x7f0824a47f8d in Gui::View3DInventorViewer::renderToFramebuffer(QOpenGLFramebufferObject*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x1ed
#15  0x7f0824a48663 in Gui::View3DInventorViewer::imageFromFramebuffer(int, int, int, QColor const&, QImage&) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x113
#16  0x7f08246cd699 in Gui::Thumbnail::SaveDocFile(Base::Writer&) const from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0xa9
#17  0x7f08237696a0 in Base::ZipWriter::writeFiles() from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADBase.so+0x150
#18  0x7f0823a932f2 in App::Document::save(Base::Writer&, bool) const from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADApp.so+0x112
#19  0x7f0823a9754a in App::Document::saveToFile(char const*) const from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADApp.so+0x47a
#20  0x7f0823a983f4 in App::Document::save() from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADApp.so+0x1f4
#21  0x7f0823af79c4 in App::DocumentPy::save(_object*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADApp.so+0x44
#22  0x7f0823af7ae7 in App::DocumentPy::staticCallback_save(_object*, _object*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADApp.so+0x57
#23  /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libpython3.8.so.1.0(PyCFunction_Call+0xf5) [0x7f0823fc2725]
#24  /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x230) [0x7f0823f6a550]
#25  /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x4ddd) [0x7f0823fd428d]
#26  /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x2ea) [0x7f0823f6bd9a]
#27  /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libpython3.8.so.1.0(PyEval_EvalCodeEx+0x3a) [0x7f0823ff940a]
#28  /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libpython3.8.so.1.0(PyEval_EvalCode+0x1c) [0x7f0823ff942c]
#29  /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libpython3.8.so.1.0(+0x1e05aa) [0x7f082401e5aa]
#30  /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libpython3.8.so.1.0(+0x224225) [0x7f0824062225]
#31  /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libpython3.8.so.1.0(PyRun_StringFlags+0x80) [0x7f08240622c0]
#32  0x7f08236f754c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADBase.so+0x6c
#33  0x7f0824705f61 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x91
#34  0x7f0824706101 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x101
#35  0x7f082467f60b in Gui::Document::save() from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x2db
#36  0x7f0824b60fd9 in Gui::MDIView::onMsg(char const*, char const**) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0xb9
#37  0x7f0824a325a9 in Gui::View3DInventor::onMsg(char const*, char const**) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x309
[... cut ...]
#72  0x7f08229d7019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libQt5Core.so.5+0x187
#73  0x7f08229dae8f in QCoreApplication::exec() from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib/libQt5Core.so.5+0xfd
#74  0x7f082463696e in Gui::Application::runApplication() from /tmp/.mount_FreeCAilS7Wf/usr/bin/../lib64/libFreeCADGui.so+0x40e
#75  /tmp/.mount_FreeCAilS7Wf/usr/bin/FreeCADLink(+0x44bf) [0x55d4be8104bf]
#76  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f08224c70b3]
#77  /tmp/.mount_FreeCAilS7Wf/usr/bin/FreeCADLink(+0x47d9) [0x55d4be8107d9]
```

tomster, 2021-03-07:
and another one. this time again, while saving
```
Cannot find icon: 0x7f98677a2760
<PartDesign> CommandBody.cpp(148): Ignore selection with no solid shape: bench#Assembly.
Recompute failed! Please check report view.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f9872cff210]
#1  0x7f98728bd257 in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/../lib/libCoin.so.80+0x137
#2  0x7f9872693149 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#3  0x7f987269ad96 in SoGLRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/../lib/libCoin.so.80+0x186
#4  0x7f987519a014 in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x64
#5  0x7f9872693ac2 in SoAction::apply(SoPathList const&, int) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/../lib/libCoin.so.80+0x5b2
#6  0x7f98751977df in Gui::SoBoxSelectionRenderActionP::apply(Gui::SoBoxSelectionRenderAction*, SoPathList const&, int) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x10f
#7  0x7f9875197859 in Gui::SoBoxSelectionRenderAction::apply(SoPathList const&, int) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x29
#8  0x7f987269a5d9 in SoGLRenderActionP::renderSingle(SoNode*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/../lib/libCoin.so.80+0x279
#9  0x7f987269aaae in SoGLRenderActionP::render(SoNode*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/../lib/libCoin.so.80+0x11e
#10  0x7f987269acb2 in SoGLRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/../lib/libCoin.so.80+0xa2
#11  0x7f987519a014 in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x64
#12  0x7f9872693e2a in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/../lib/libCoin.so.80+0x2ea
#13  0x7f9875197bca in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x2a
#14  0x7f9875260f8d in Gui::View3DInventorViewer::renderToFramebuffer(QOpenGLFramebufferObject*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x1ed
#15  0x7f9875261663 in Gui::View3DInventorViewer::imageFromFramebuffer(int, int, int, QColor const&, QImage&) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x113
#16  0x7f9874ee6699 in Gui::Thumbnail::SaveDocFile(Base::Writer&) const from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0xa9
#17  0x7f9873f826a0 in Base::ZipWriter::writeFiles() from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADBase.so+0x150
#18  0x7f98742ac2f2 in App::Document::save(Base::Writer&, bool) const from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADApp.so+0x112
#19  0x7f98742b054a in App::Document::saveToFile(char const*) const from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADApp.so+0x47a
#20  0x7f98742b13f4 in App::Document::save() from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADApp.so+0x1f4
#21  0x7f98743109c4 in App::DocumentPy::save(_object*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADApp.so+0x44
#22  0x7f9874310ae7 in App::DocumentPy::staticCallback_save(_object*, _object*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADApp.so+0x57
#23  /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib/libpython3.8.so.1.0(PyCFunction_Call+0xf5) [0x7f98747db725]
#24  /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x230) [0x7f9874783550]
#25  /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x4ddd) [0x7f98747ed28d]
#26  /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x2ea) [0x7f9874784d9a]
#27  /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib/libpython3.8.so.1.0(PyEval_EvalCodeEx+0x3a) [0x7f987481240a]
#28  /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib/libpython3.8.so.1.0(PyEval_EvalCode+0x1c) [0x7f987481242c]
#29  /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib/libpython3.8.so.1.0(+0x1e05aa) [0x7f98748375aa]
#30  /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib/libpython3.8.so.1.0(+0x224225) [0x7f987487b225]
#31  /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib/libpython3.8.so.1.0(PyRun_StringFlags+0x80) [0x7f987487b2c0]
#32  0x7f9873f1054c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADBase.so+0x6c
#33  0x7f9874f1ef61 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x91
#34  0x7f9874f1f101 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x101
#35  0x7f9874e9860b in Gui::Document::save() from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x2db
#36  0x7f9875379fd9 in Gui::MDIView::onMsg(char const*, char const**) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0xb9
#37  0x7f987524b5a9 in Gui::View3DInventor::onMsg(char const*, char const**) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x309
#38  0x7f9874e290dd in Gui::Application::sendMsgToActiveView(char const*, char const**) from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x3d
[... cut ...]
#74  0x7f9874e4f96e in Gui::Application::runApplication() from /tmp/.mount_FreeCAQVSv5O/usr/bin/../lib64/libFreeCADGui.so+0x40e
#75  /tmp/.mount_FreeCAQVSv5O/usr/bin/FreeCADLink(+0x44bf) [0x55e7d4c534bf]
#76  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f9872ce00b3]
#77  /tmp/.mount_FreeCAQVSv5O/usr/bin/FreeCADLink(+0x47d9) [0x55e7d4c537d9]
```
i started the model from scratch (this time without copying anything over, i'm attaching the file. again, this is using 2021.0307
[bench2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6097819/bench2.zip)

realthunder, 2021-03-08:
Thanks for your diligent report, I have finally identified a tricky crash reported in #486. At least I hope that's the one (fingers crossed). I'll make a new release soon.

tomster, 2021-03-08:
I gave the 2021.03.08 release a quick spin with my most troublesome model and try as i might i could not get it to crash, so this looks promising, yay!
i will keep the issue open for a bit to do some more testing and will report back.
thank you so much @realthunder for responding so quickly and helpfully!

tomster, 2021-03-08:
spent nearly two hours now working on that model without a single crash. thanks again. also, i’ve just become a patreon supporter of you. nothing big tier wise but i shall keep it running as long as i use freecad. so hopefully indefinitely :-)

closed 2021-03-08 by tomster