realthunder/FreeCAD_assembly3 #490: Tape => Taper

bf5man, 2021-03-04:
Small detail, in the Part Design->Pocket operation, the "tape" parameters should probably renamed "taper" entries since that's the proper name other commercial CAD tool use to refer to tapered faces.
- Tape angle=>Taper angle
- 2nd tape angle=>2nd taper angle
- Inner tape angle=>Inner taper angle
- 2nd inner tape angle=>2nd inner taper angle

Anyway, this new feature is really nice, thanks RT!

realthunder, 2021-03-04:
Oops, nice catch. Thanks!

bf5man, 2021-03-04:
It got me thinking, I'm also not totally right. 

The "taper" term is correct, but other CAD softwares refer to this feature as "draft angle":

- Onshape: https://cad.onshape.com/help/Content/draft.htm
- Solidworks: http://help.solidworks.com/2019/english/SolidWorks/sldworks/t_parting_line_draft.htm?format=P&value=
- SolidEdge: https://docs.plm.automation.siemens.com/tdoc/se/2020/se_help/#uid:index#goto:create_part:feat3c

In fact, Freecad even has a Draft face feature.

So maybe this would benefit from a consensus with more users.

realthunder, 2021-03-04:
Well, I think we should keep the `Taper` naming just to avoid confusion with the `Draft` feature.

closed 2021-06-12 by oficinerobotica