realthunder/FreeCAD_assembly3 #491: issues in PartDesignLinkBranch

jpg87Fr, 2021-03-05:
I pointed them out in this post :
https://forum.freecadweb.org/viewtopic.php?f=20&t=56295

[image]
[image]

[lv5_jpg87_LinkBranch.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6090031/lv5_jpg87_LinkBranch.zip)

ceremcem, 2021-03-05:
Is there any particular reason why you don't want to click the "toggle visibility" icon: 

https://user-images.githubusercontent.com/6639874/110120450-1d1b5e80-7dce-11eb-88ef-e9d8b6e63ec5.mp4

ceremcem, 2021-03-05:
Sorry, the visibility icon can only solve 1 object's visibility issue. Please ignore my previous post.

realthunder, 2021-03-06:
The Fillet, actually all dress up features (Draft, Chamfer), can only belong to the same solid group of its base. Since you created a new solid when making those two socket, and Fillet is applied on the Mirror which is in another solid, that's why the sockets are not there.

There is indeed a bug here nevertheless. When created the Fillet, it should be auto inserted after the Mirror, instead of at the bottom. I have fixed this problem.

If you want those two socket, then you should delete the Fillet, and turn off `New solid` option in `SlideScrewReceiver`. The recreate the Fillet. With the bug fixed, you won't need to delete the Fillet in the future.

oficinerobotica, 2021-06-12:
@realthunder would you consider this issue fixed? can it be closed?

luzpaz, 2021-07-17:
@jpg87Fr please test and report back

luzpaz, 2021-07-30:
Closing this ticket, @jpg87Fr  
Please re-open if merited.

closed 2021-07-30 by luzpaz