realthunder/FreeCAD_assembly3 #492: Model Variant Initialization through dropdown

ceremcem, 2021-03-06:
@realthunder  I barely remember that I saw an example of object variant initialization through a dropdown within one of your forum posts, am I right?

You were creating a link for an external object and manipulate that link (like, assigning predefined arguments) through a dropdown within the properties. Can you point the reference post please?

realthunder, 2021-03-06:
http://forum.freecadweb.org/viewtopic.php?f=17&t=42183

Someone has also made a video about this https://www.youtube.com/watch?v=PqARo51D4E0&t=1s

closed 2021-03-06 by ceremcem

ceremcem, 2021-06-22:
A very nice summary from @bgamari: https://github.com/realthunder/FreeCAD_assembly3/issues/685#issue-926252104