realthunder/FreeCAD_assembly3 #495: Sketcher: mouse selection of constraints behaves weirdly

wohltat, 2021-03-06:
often it is not possible to select the constraints under the mouse cursor.

Here is a small example:
When trying to select one of the two constraints, even the other one is chosen. Sometimes it not possible at all to click on the constraint.
[image]

-------------

OS: Linux Mint 19 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0301.23964 +3306 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 32f78691fae500e494879493abc2b74cff9585d3
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United Kingdom (en_GB)

realthunder, 2021-03-06:
Try the same trick in [#497](https://github.com/realthunder/FreeCAD_assembly3/issues/497#issuecomment-791901475)

herrgerd, 2021-03-07:
> 
> 
> Try the same trick in [#497](https://github.com/realthunder/FreeCAD_assembly3/issues/497#issuecomment-791901475)

That "trick" is a nice feature in FreeCAD. But I'm still with the OP saying, that selecting constraints (especially many next to each other) does not work really well... It should just work without a feature.

Other CAD packages have similar issues, one solution is to give you a drop-down with all the constraints in an area X around the cursor.

Rabbit-sk, 2021-03-08:
> one solution is to give you a drop-down with all the constraints in an area X around the cursor.

I agree, it would be a perfect solution.

wohltat, 2021-07-09:
> Try the same trick in [#497](https://github.com/realthunder/FreeCAD_assembly3/issues/497#issuecomment-791901475)

Actually that trick does not help. With G,G  you can only choose between the constraints that is already selected and not the one you are hovering over (the one you want to select).