realthunder/FreeCAD_assembly3 #499: Can not import svg symbol into techdraw drawing

ngocminhdao88, 2021-03-06:
I could not import my custom svg symbol into a techdraw drawing

Python console and report output:
`>>> ### Begin command TechDraw_Symbol
>>> f = open("/home/minh/Workspace/taper-buchse/taper.svg",'r')
>>> svg = f.read()
>>> f.close()
>>> App.getDocument('taper_buchse').getObject('Page').addObject('TechDraw::DrawViewSymbol','Symbol')`

`08:25:44  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'AttributeError'>: <no object bound to built-in method addObject>`

My file with a svg symbol, that i want to import:
[taper.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6094925/taper.zip)

My system:
OS: Ubuntu 20.04.2 LTS (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0301.23964 +3306 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 32f78691fae500e494879493abc2b74cff9585d3
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

ceremcem, 2021-03-06:
I can confirm that I could reproduce the issue and above commit fixed it.

luzpaz, 2022-01-12:
Tried recreating this bug. Results:

```
>>> f = open("/home/foo/Downloads/taper/taper.svg",'r')
>>> svg = f.read()
>>> f.close()
>>> App.getDocument('taper_buchse').getObject('Page').addObject('TechDraw::DrawViewSymbol','Symbol')`
  File "<input>", line 1
    App.getDocument('taper_buchse').getObject('Page').addObject('TechDraw::DrawViewSymbol','Symbol')`
                                                                                                    ^
SyntaxError: invalid syntax
```

```
OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

realthunder, 2022-01-12:
> Tried recreating this bug. Results:

You mean you still have this problem with the latest release? That's odd. I can't reproduce it.

luzpaz, 2022-01-12:
It's a different issue. Lemme check again

luzpaz, 2022-03-21:
@realthunder I can still reproduce
1. Download the zip
2. extract and get the path to the `.svg` file
3. fire up FC and drop the .FCStd file in to FC
4. switch to the TechDraw workbench
5. in the python console invoke the script
    ```python
     f = open("/path/to/taper.svg",'r')
    svg = f.read()
    f.close()
    App.getDocument('taper_buchse').getObject('Page').addObject('TechDraw::DrawViewSymbol','Symbol')
    ```
Result: 
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
AttributeError: <no object bound to built-in method addObject>
```

realthunder, 2022-03-22:
That's strange. Can you please record a screen cast showing the problem? Are clicking the following button? Or do you mean you paste the script to console? The script is wrong, and I've fixed it.

[image]

luzpaz, 2022-03-22:
I was using the script per OP in https://github.com/realthunder/FreeCAD_assembly3/issues/499#issue-823586768

closed 2022-07-25 by luzpaz