realthunder/FreeCAD_assembly3 #500: Different behaviour between windows and linux

j-wiedemann, 2021-03-06:
Hi @realthunder we are working on a file: https://github.com/nadarbreicq/Barbatronic/blob/master/boxGenerator/boxGeneratorV1.3.FCStd originally made with this fc version :
```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0301.23964 +3306 (Git)
Build type: Release
Branch: LinkStage3
Hash: 32f78691fae500e494879493abc2b74cff9585d3
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: French/France (fr_FR)
```
If I open it with this version:
```
OS: Ubuntu 20.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0301.23964 +3306 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 32f78691fae500e494879493abc2b74cff9585d3
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: French/France (fr_FR)
```
and changing a parameter (like first one Depth) in Parameters object, I get an error:
```
<Exception> TopoShapeEx.cpp(936): Invalid shape name ?Face6
20:34:17  Recompute failed! Please check report view.
```
On Ubuntu if you mark the doc to recompute before doing anything there is no errors.
If you save the doc on Ubuntu then send it back to windows there is the same error now on windows.

So I'm thinking of an imcompatibility between OCC version. Do you think the same ? Do you think we will soon get same OCC version between windows and linux and mac ? Do you need more informations and investigation if you think this need to be compatible between os and occ version ?

ceremcem, 2021-03-06:
Changing `Depth` parameter does not cause any exception on my side in version: 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27270 (Git)
Build type: Release
Branch: LinkStage3
Hash: 32f78691fae500e494879493abc2b74cff9585d3
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2021-03-06:
I think it is most likely OCCT version problem. There is some mentioning about OCCT 7.5 conda build progress in [this](https://forum.freecadweb.org/viewtopic.php?f=10&t=55077&start=140#p481982) thread.

luzpaz, 2022-07-01:
@j-wiedemann can you retest on latest Daily (it uses OCC 7.6.2 AFAIU)