realthunder/FreeCAD_assembly3 #503: On dialogs buttons only clicks over the text take effect

tonilupi, 2021-03-07:
Hi, first of all I want to thank you for the great effort you're making with your Linkbranch.

In my first contribution I want to show you this issue: On the buttons of the dialog boxes it is necessary to click on the texts for the action to be take in consideration.

[image]

On red zone nothing happen when you press, only on green zone the pushing take effect.

ceremcem, 2021-03-07:
Duplicate of https://github.com/realthunder/FreeCAD_assembly3/issues/479

tonilupi, 2021-03-07:
> Duplicate of #479

Sorry for the inconvenience. Can I erase the issue or I must to close it?

ceremcem, 2021-03-07:
No problem. You can just close it.

closed 2021-03-07 by tonilupi