realthunder/FreeCAD_assembly3 #504: problems compiling on ubuntu 20.04

wohltat, 2021-03-08:
The FreeCAD/FreeCAD repository seems to compile just fine.
I used the [freecad-daily PPA](https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily).

When trying to cmake i got the error below. As it seems the linkstage3 branch still uses qt4. The qt4 packages are not part of the ubuntu standard repositories and in the PPA there seem to be not all necessary packages related to qt4 (like shiboken). 

Will there be a switch to qt5 in the near future to match the mainline FreeCAD?
How is linkstage3 meant to be compiled?
Am i missing / overlook something here, i.e. is there an easy way to make it compile? 

```
CMake Warning at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:122 (find_package):
  By not providing "FindShiboken.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Shiboken",
  but CMake did not find one.

  Could not find a package configuration file provided by "Shiboken" with any
  of the following names:

    ShibokenConfig.cmake
    shiboken-config.cmake

  Add the installation prefix of "Shiboken" to CMAKE_PREFIX_PATH or set
  "Shiboken_DIR" to a directory containing one of the above files.  If
  "Shiboken" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:75 (SetupShibokenAndPyside)

CMake Error at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:124 (message):
  ===================

  shiboken not found.

  ===================

Call Stack (most recent call first):
  CMakeLists.txt:75 (SetupShibokenAndPyside)

```

Compiling shiboken manually also fails btw, with following error. Looks pretty outdated. 
```
>>> make -j 3                                                                                                                                                        ‹git:master ✔› 08:49.37 Mon Mar 08 2021 >>> 
[  1%] Built target libminimal
[  5%] Built target libshiboken
[  5%] Building CXX object tests/libsample/CMakeFiles/libsample.dir/size.cpp.o
[  6%] Building CXX object tests/libsample/CMakeFiles/libsample.dir/simplefile.cpp.o
[ 16%] Built target apiextractor
[ 17%] Generating testmodifydocumentation.moc
Scanning dependencies of target testmodifydocumentation
[ 17%] Building CXX object ApiExtractor/tests/CMakeFiles/testmodifydocumentation.dir/testmodifydocumentation.cpp.o
[ 17%] Generating testtyperevision.moc
Scanning dependencies of target testtyperevision
[ 18%] Building CXX object ApiExtractor/tests/CMakeFiles/testtyperevision.dir/testtyperevision.cpp.o
/home/username/build/Shiboken/tests/libsample/simplefile.cpp: In member function ‘bool SimpleFile::exists() const’:
/home/username/build/Shiboken/tests/libsample/simplefile.cpp:93:12: error: cannot convert ‘std::ifstream’ {aka ‘std::basic_ifstream<char>’} to ‘bool’ in return
   93 |     return ifile;
      |            ^~~~~
/home/username/build/Shiboken/tests/libsample/simplefile.cpp: In static member function ‘static bool SimpleFile::exists(const char*)’:
/home/username/build/Shiboken/tests/libsample/simplefile.cpp:100:12: error: cannot convert ‘std::ifstream’ {aka ‘std::basic_ifstream<char>’} to ‘bool’ in return
  100 |     return ifile;
      |            ^~~~~
make[2]: *** [tests/libsample/CMakeFiles/libsample.dir/build.make:563: tests/libsample/CMakeFiles/libsample.dir/simplefile.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1614: tests/libsample/CMakeFiles/libsample.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 18%] Linking CXX executable testmodifydocumentation
[ 18%] Linking CXX executable testtyperevision
[ 18%] Built target testmodifydocumentation
[ 18%] Built target testtyperevision
make: *** [Makefile:160: all] Error 2

```

ceremcem, 2021-03-08:
Would you like to try https://github.com/ceremcem/build-freecad-asm3 out? I would recommend the default (LXC based) procedure but you can of course stick to [Building directly on host](https://github.com/ceremcem/build-freecad-asm3#building-directly-on-host) part.

wohltat, 2021-03-08:
First i tried building on host. The commands do not work as is.
```
$ git clone https://github.com/ceremcem/build-freecad-asm3
$ ./build-freecad-asm3/build.sh 

```
With a minor modification the packages.txt is found:
```
$ git clone https://github.com/ceremcem/build-freecad-asm3
$ cd build-freecad-asm3/
$ ./build.sh 
```

Building directly on host failed with
`E: Unable to locate package python3-pyside2uic`

i then removed that package from packages.txt and tried again. Then it fails silently with 
```
-------------------------------
Building in ../fc-build/Release
-------------------------------
```
---------------
trying the lxc path, i got stuck at the following point after a long time setting up lxd / lxc and trying a lot of things:

```
>>> sudo lxc-create -n fc -t debian -- -r buster --packages xbase-clients nano sudo tmux git                                                                                                  19:30.45 Mon Mar 08 2021 >>> 
[sudo] password for user: 
debootstrap is /usr/sbin/debootstrap
Checking cache download in /var/cache/lxc/debian/rootfs-buster-amd64 ... 
gpg: key 7638D0442B90D010: 4 signatures not checked due to missing keys
gpg: key 7638D0442B90D010: "Debian Archive Automatic Signing Key (8/jessie) <ftpmaster@debian.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
Downloading debian minimal ...
I: Retrieving InRelease 
I: Checking Release signature
E: Release signed by unknown key (key id DCC9EFBF77E11517)
   The specified keyring /var/cache/lxc/debian/archive-key.gpg may be incorrect or out of date.
   You can find the latest Debian release key at https://ftp-master.debian.org/keys.html
Failed to download the rootfs, aborting.
Failed to download 'debian base'
failed to install debian
lxc-create: fc: lxccontainer.c: create_run_template: 1620 Failed to create container from template
lxc-create: fc: tools/lxc_create.c: main: 319 Failed to create container fc

```

ceremcem, 2021-03-08:
Let's handle `build-freecad-asm3` related issue in its own issue tracker. I copied your issue to: https://github.com/ceremcem/build-freecad-asm3/issues/11

wohltat, 2021-03-08:
build-fc.sh:110 only works if the folder already exists, otherwise build_dir (new) will be empty even if build_dir (old) was not empty (build_dir was '../fc-build'):
```
build_dir=$(readlink -f $build_dir)

mkdir -p $build_dir
```
i manually followed the build.sh script and now it starts compiling.
I think i did a mistake which revealed something confusing. I was on the "realthunder/FreeCAD_assembly3:master" branch instead of the "realthunder/FreeCAD:LinkStage3" branch. This in itself is confusing because the release of FreeCAD:LinkStage3 is on FreeCAD_assembly3. Furthermore in your build script there is to read "Assembly3" and "asm3" which adds to the confusion.

Why are there these two repos in the first place?

ceremcem, 2021-03-08:
> Why are there these two repos in the first place?

This repo started its life as an alternative to the Assembly3/Wiki instructions. As I'm not authorized to add it to any repository, it exists as a separate one. 

> build-fc.sh:110 only works if the folder already exists, otherwise build_dir (new) will be empty even if build_dir (old) was not empty (build_dir was '../fc-build'):

Thanks, I fixed it. 

> This in itself is confusing because the release of FreeCAD:LinkStage3 is on FreeCAD_assembly3

Confusion comes from the history: Back in the days we had many place to track issues, like FreeCAD forum, LinkStage3/issues, Assembly3/issues. In the first days issues were tracked via the forum entry. It was very complex to track the issues there, so we started using Assembly3/issues for Assembly3 related questions. Since build instructions was on Assembly3/README, I also posted many build related question the Assembly3/issues. Since I can't know whether exact reason for an issue is related to Assembly3 WB or LinkStage3, I kept posting every question to Assembly3/issues. 

Then some folks started posting issues to LinkStage3/issues when they were exactly sure that the issue is related to LinkStage3. So we started tracking both trackers. I don't know if anyone still posting issues to that forum entry.

So, *I guess* because of both Assembly3/issues was the main issue tracker and the build instructions was on Assembly3/README, realthunder decided to put the releases to Assembly3 repo. 

> Furthermore in your build script there is to read "Assembly3" and "asm3" which adds to the confusion

I may not quite get this. What do you mean by that?

wohltat, 2021-03-10:
Btw the `shiboken not found.` error was solved by the cmake flag -DBUILD_QT5=ON
shiboken is for Qt4

luzpaz, 2021-08-04:
Is this ticket still pending ?

ceremcem, 2021-08-24:
@luzpaz This ticket is resolved: https://github.com/ceremcem/build-freecad-asm3/issues/11#issuecomment-803000634

closed 2021-08-24 by ceremcem