realthunder/FreeCAD_assembly3 #505: Grid not fully displayed in 3D

MarcelP102, 2021-03-08:
When enabling the grid in 3D it's not fully displayed. See screenshot:
[image]

My system:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0307.24301 +3357 (Git)
Build type: Release
Branch: LinkStage3
Hash: 289a25b99096421a8f475088e478d4bba9d1a37b
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Dutch/Netherlands (nl_NL)

snelweg, 2021-03-10:
I thought it was a feature to be honest. The smaller grid looks way better in my opinion.

realthunder, 2021-03-10:
Eh...It's not a feature. But the `View fit` command will ignore the grid, and that's a feature. The grid will be shown in full if you manually zoom out with [this](ca806e636c2e24b270d0f1d065e292f82d953420) commit.

MarcelP102, 2021-03-11:
Confirm it's fixed, thanks.

closed 2021-03-11 by MarcelP102