realthunder/FreeCAD_assembly3 #506: Start page error message about missing thumbnail file

adrianinsaval, 2021-03-09:
Start page shows an annoying error message about missing thumbnails:
[image]
[image]

The message appears many times with different file names. If I leave the start page open the messages randomly appears again after some time, I don't know what triggers this. FreeCAD master shows no errors.
I probably wiped the temp folder before that but I don't really remember.

adrianinsaval, 2021-03-09:
Just noticed one of the messages complains about a missing image from start workbench not just another thumbnail from temp:
[image]
That one I definitively didn't delete.

realthunder, 2021-03-09:
I've also seen this type of error some time ago. But it goes away before I got a chance to inspect it. And I can't reproduce it anymore. The above referenced commit simply change the warning dialog into a report view log message, just in case it happens again.

Meanwhile, as a work around, you can manually remove all MRU entries in (BaseApp/Preferences/RecentFiles) and set `RecentFiles` to 0, using parameter editor. Then restart FreeCAD and see if recent file list has been cleared.

luzpaz, 2021-08-05:
@realthunder can this be closed?

realthunder, 2021-08-05:
Yeah, most likely fixed by [this](https://github.com/realthunder/FreeCAD/commit/16d876c8eea7c34d91b589626deaf222896e9c60).

closed 2021-08-05 by realthunder