realthunder/FreeCAD_assembly3 #508: Question: TreeView transparent mode in 2D view

ngocminhdao88, 2021-03-09:
I am using FreeCAD with overlaid treeview in transparent mode. Everything works file in 3D view, i can see objects through the treeview. However this mode doesn't work that great in 2D view. Mouseover brings the treeview's background to front (dark gray in my case), which doesn't look great and kind of annoying.

Is there a way to fix it? Can disabling treeview mouseover event help to improve user's experience?

[image]

realthunder, 2021-03-10:
Can you please post your FreeCAD About information here? Also please post your link.user.cfg file. You can send to my email realthunder.dev at gmail. Depending on your platform, on Windows the file is at `c:\users/<your_user_name>/AppData/Roaming/FreeCAD/link.user.cfg`. On Linux, it is in `~/.FreeCAD/link.user.cfg`. On Mac it is in `~/Library/Preferences/FreeCAD/link.user.cfg`.

ngocminhdao88, 2021-03-10:
Hi realthunder,

here is information about my FreeCAD.

OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0307.24301 +3357 (Git)
Build type: Release
Branch: LinkStage3
Hash: 289a25b99096421a8f475088e478d4bba9d1a37b
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

and here is my user configuration
[link.user.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6114271/link.user.zip)

realthunder, 2021-03-11:
Hi, just checked your setting. Is there any particular reason of why you disabled `Auto hide in non 3D view` in `Preference -> Display -> UI`? In 3D view, the overlay has an option to `Toggle transparent mode`, such that the overlay will change to opaque when you mouse over it. But in non 3D view, this option is ignored, and the overlay will always become opaque when mouse over.

The reason being that I thought that overlay mode in non 3D view is generally unusable, and making it opaque can at least make it a bit more bearable. Also because that you cannot _Pan_ a non 3D view, the overlay will obscure the underlying widgets and making them inaccessible, unless of course you have auto hide enabled, in which case, the mouse over transparency change will be a no issue as well.

ngocminhdao88, 2021-03-11:
Hi realthunder,

i am using TechDraw to generate drawing for manufacturing parts. Using TechDraw without tree view is really awkward.

 In TechDraw you can pan and zoom, but not in SpreadSheet. I can see what you describe, that tree view blocks cells below it. But i am mostly undocking SpreadSheet and let it float when i design 3D objects parametrically.

Here is how my workspace look like.
[image]

Using tree view in non transparent mode reduces my screen real estate for 3D object a bit too much and i want the properties always there to quickly modify things. Of course i can using only combo view showing both model tree and properties. However, vertical space is scarce with modern monitors these days.

Can i make the tree view background always transparent or always opaque in 2D view? This can reduce the screen flashing issue, which i described in first post.

oficinerobotica, 2021-06-06:
The issue isn't present anymore in version:
```
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```

closed 2021-06-06 by oficinerobotica