realthunder/FreeCAD_assembly3 #509: Segfault on Mac in Sketcher

theory14, 2021-03-09:
When editing in the Sketcher (usually invoked from Part Design), I get frequent segfaults when either trying to close the sketch or save the project.  This seems to happen as the mouse crosses out of the 3D View area.

Crash report details below:

```
Process:               FreeCADLink [28999]
Path:                  /Users/USER/*/FreeCAD-2021-0308.app/Contents/Resources/bin/FreeCADLink
Identifier:            ???
Version:               ??? (0.19)
Code Type:             X86-64 (Native)
Parent Process:        ??? [28995]
User ID:               501

Date/Time:             2021-03-09 08:07:59.667 -0500
OS Version:            macOS 11.2.2 (20D80)
Report Version:        12
Anonymous UUID:        F5385699-6C03-4B88-2720-694A767DA345

Sleep/Wake UUID:       49CED2A9-6230-42DA-A3BE-AF5D0E742B4A

Time Awake Since Boot: 220000 seconds
Time Since Wake:       800 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [28999]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libQt5Widgets.5.12.1.dylib    	0x000000010a440e17 0x10a432000 + 60951
1   libQt5Widgets.5.12.1.dylib    	0x000000010a4a645a 0x10a432000 + 476250
2   libFreeCADGui.dylib           	0x0000000108d2c21d Gui::OverlayManager::Private::onTimer() + 1101
3   libQt5Core.5.12.1.dylib       	0x000000010ae3cc09 QMetaObject::activate(QObject*, int, int, void**) + 1833

Thread 1:: com.apple.CFSocket.private
0   libsystem_kernel.dylib        	0x00007fff203081fe __select + 10
1   com.apple.CoreFoundation      	0x00007fff204536c3 __CFSocketManager + 644
2   libsystem_pthread.dylib       	0x00007fff20334950 _pthread_start + 224
3   libsystem_pthread.dylib       	0x00007fff2033047b thread_start + 15

Thread 2:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff202ffe7e mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff203001f0 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff2042cbf7 __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff2042b2ca __CFRunLoopRun + 1315
4   com.apple.CoreFoundation      	0x00007fff2042a6ce CFRunLoopRunSpecific + 563
5   com.apple.AppKit              	0x00007fff22dd07c6 _NSEventThread + 124
6   libsystem_pthread.dylib       	0x00007fff20334950 _pthread_start + 224
7   libsystem_pthread.dylib       	0x00007fff2033047b thread_start + 15

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff202ffe7e mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff203001f0 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff2042cbf7 __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff2042b2ca __CFRunLoopRun + 1315
4   com.apple.CoreFoundation      	0x00007fff2042a6ce CFRunLoopRunSpecific + 563
5   com.apple.CoreFoundation      	0x00007fff204b0cc9 CFRunLoopRun + 40
6   com.3dconnexion.driver.client 	0x000000010baa26ad -[ConnexionClientAPI clientThreadEntry] + 220
7   com.apple.Foundation          	0x00007fff211b0e1d __NSThread__start__ + 1042
8   libsystem_pthread.dylib       	0x00007fff20334950 _pthread_start + 224
9   libsystem_pthread.dylib       	0x00007fff2033047b thread_start + 15

Thread 4:
[... 696 lines cut ...]
__TEXT                           647.1M      613 
__UNICODE                          588K        1 
mapped file                       78.9M       22 
shared memory                       40K        4 
===========                     =======  ======= 
TOTAL                              4.3G     2494 
TOTAL, minus reserved VM space     2.4G     2494 

Model: iMac19,1, BootROM 1554.80.3.0.0, 8 processors, 8-Core Intel Core i9, 3.6 GHz, 32 GB, SMC 2.46f12
Graphics: kHW_AMDRadeonProVega48Item, Radeon Pro Vega 48, spdisplays_pcie_device, 8 GB
Memory Module: BANK 0/ChannelA-DIMM0, 16 GB, DDR4, 2667 MHz, Kingston, 9905712-009.A00G
Memory Module: BANK 2/ChannelB-DIMM0, 16 GB, DDR4, 2667 MHz, Kingston, 9905712-009.A00G
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Sep 11 2020 16:57:49 version 9.30.440.2.32.5.61 FWID 01-129bddb
Bluetooth: Version 8.0.3d9, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
USB Device: USB 3.1 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB-C ADAPTER VGA
USB Device: Back-UPS RS 1500 LCD FW:839.H7 .D USB FW:H7
USB Device: CadMouse Pro Wireless Left
USB Device: USB 2.0 Hub [MTT]
USB Device: Portable Super Multi Drive
USB Device: EPSON Scanner
Thunderbolt Bus: iMac, Apple Inc., 47.4
```

theory14, 2021-03-20:
This also occurs with 03-11.  I can provide another dump for that version if needed.  I did a similar design with upstream (0.19-24267 from 2021/02/26 and didn't have the crash problem.

realthunder, 2021-03-20:
Is the dump similar? Just compare the crashed thread stack

> Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
> 0   libQt5Widgets.5.12.1.dylib    	0x000000010a440e17 0x10a432000 + 60951
> 1   libQt5Widgets.5.12.1.dylib    	0x000000010a4a645a 0x10a432000 + [...]

If the crash is repeatable, please also provide a screencast showing the sequence, along with your user config file at `~/Library/Preferences/FreeCAD/link.user.cfg`. Thanks

theory14, 2021-03-22:
Will provide the requested info as soon as I can.  Will also include the file that results in the issue.  I can go for a while without hitting this and then once I hit it on a particular sketch, it takes many attempts to successfully compete the sketch.

theory14, 2021-04-03:
Finally got some time to do some fire up FreeCAD.  Editing a sketch, I got another crash.  Looks to be the same as before.

```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libQt5Widgets.5.12.1.dylib    	0x000000010f8dbe17 0x10f8cd000 + 60951
1   libQt5Widgets.5.12.1.dylib    	0x000000010f94145a 0x10f8cd000 + 476250
2   libFreeCADGui.dylib           	0x000000010e1bf0dd Gui::OverlayManager::Private::onTimer() + 1101
3   libQt5Core.5.12.1.dylib       	0x00000001102d0c09 QMetaObject::activate(QObject*, int, int, void**) + 1833
```

My link.user.cfg file:

```
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<FCParameters>
  <FCParamGroup Name="Root">
    <FCParamGroup Name="BaseApp">
      <FCParamGroup Name="LogLevels">
        <FCInt Name="Default" Value="2"/>
      </FCParamGroup>
      <FCParamGroup Name="Preferences">
        <FCParamGroup Name="General">
          <FCText Name="FileOpenSavePath">/Users/csgordon/Documents/CAD/FreeCAD_Learning</FCText>
          <FCText Name="AutoloadModule">PartDesignWorkbench</FCText>
          <FCBool Name="ShowSplasher" Value="1"/>
          <FCBool Name="PythonWordWrap" Value="1"/>
          <FCInt Name="ToolbarIconSize" Value="24"/>
          <FCText Name="Language">English</FCText>
          <FCText Name="FileExportFilter">OpenSCAD Format (*.scad)</FCText>
          <FCText Name="PreferenceGeometry">767 149 1512 1559</FCText>
          <FCText Name="LastModule">PartDesignWorkbench</FCText>
          <FCBool Name="RecentIncludesExported" Value="0"/>
          <FCBool Name="ConfirmAll" Value="0"/>
        </FCParamGroup>
        <FCParamGroup Name="Units">
          <FCInt Name="UserSchema" Value="1"/>
          <FCInt Name="Decimals" Value="4"/>
          <FCInt Name="FracInch" Value="8"/>
          <FCInt Name="DecimalsPreSel" Value="-1"/>
        </FCParamGroup>
        <FCParamGroup Name="Macro">
          <FCBool Name="LocalEnvironment" Value="1"/>
          <FCText Name="MacroPath">/Users/csgordon/Library/Preferences/FreeCAD/Macro</FCText>
          <FCBool Name="RecordGui" Value="1"/>
          <FCBool Name="GuiAsComment" Value="1"/>
          <FCBool Name="ScriptToPyConsole" Value="1"/>
          <FCBool Name="ScriptToFile" Value="0"/>
          <FCText Name="ScriptFile">FullScript.FCScript</FCText>
        </FCParamGroup>
        <FCParamGroup Name="Mod">
          <FCParamGroup Name="OpenSCAD">
            <FCInt Name="useMaxFN" Value="16"/>
            <FCInt Name="exportConvexity" Value="10"/>
            <FCFloat Name="exportFa" Value="12.000000000000"/>
            <FCFloat Name="exportFs" Value="2.000000000000"/>
            <FCFloat Name="meshdeflection" Value="0.000000000000"/>
            <FCText Name="openscadexecutable"/>
            <FCBool Name="printVerbose" Value="0"/>
            <FCBool Name="useViewProviderTree" Value="0"/>
            <FCBool Name="useMultmatrixFeature" Value="0"/>
          </FCParamGroup>
          <FCParamGroup Name="Import">
            <FCParamGroup Name="hSTEP">
              <FCBool Name="ReadShapeCompoundMode" Value="1"/>
            </FCParamGroup>
            <FCBool Name="UseAppPart" Value="1"/>
            <FCBool Name="UseLegacyImporter" Value="0"/>
            <FCBool Name="UseBaseName" Value="1"/>
            <FCBool Name="ImportHiddenObject" Value="1"/>
            <FCBool Name="ExportHiddenObject" Value="1"/>
[... 1854 lines cut ...]
          <FCParamGroup Name="CurvesWorkbench"/>
          <FCParamGroup Name="TechDrawWorkbench"/>
          <FCParamGroup Name="BIMWorkbench"/>
          <FCParamGroup Name="WebWorkbench"/>
          <FCParamGroup Name="WebToolsWorkbench"/>
          <FCParamGroup Name="NoneWorkbench"/>
          <FCParamGroup Name="FemWorkbench"/>
          <FCParamGroup Name="SpreadsheetWorkbench"/>
        </FCParamGroup>
        <FCParamGroup Name="System"/>
      </FCParamGroup>
    </FCParamGroup>
    <FCParamGroup Name="Plugins">
      <FCParamGroup Name="parts_library"/>
      <FCParamGroup Name="addonsRepository">
        <FCBool Name="readWarning" Value="1"/>
      </FCParamGroup>
    </FCParamGroup>
  </FCParamGroup>

</FCParameters>

```

I'll try to get a screencast as requested.

luzpaz, 2021-06-26:
@theory14 screencast?

theory14, 2021-06-26:
Sorry I've not gotten to this.  Lots of other stuff came up (life and all) and haven't had a chance to get back to this.  Will try to grab the newest release and see if I can find the problem again.

theory14, 2021-06-26:
I just tried with 20210615 Daily with the type of sketch that would cause crashes in the past.  I didn't experience any issues.  If this appears again in the other use, I'll re-open (or open a new bug) with the screencast as requested.  Since I can't duplicate with the newer builds, I'll close this ticket.

closed 2021-06-26 by theory14