realthunder/FreeCAD_assembly3 #510: Recent files thumbnails broken

pgilfernandez, 2021-03-10:
Since a few builds the Recent Files list thumbnails get broken each time I add a new build. Right now I'm using your last one, asm3-20210301.

This is the popup I get, one popup per thumbnail and every time I save a file (while the Start page is loaded, if not it doesn't happen):
[image]

In the Recent List this is one example of what I get:
[image]

I don't know if it is something related with an accent (as you see in the last picture, my name is "Pablo Gil Fernández") or something else.

Thanks

realthunder, 2021-03-10:
Oh, my FreeCAD repo's issue page is actually more appropriate. But here also can. I'll investigate.

realthunder, 2021-03-11:
I can't reproduce the problem. Are you using MacOS native saving file dialog, something like below? Can you please try the tips [here](https://github.com/realthunder/FreeCAD_assembly3/issues/506#issuecomment-793310500) to wipe out your recent file list information and try save again?

[image]

Note that when I save the file, the Start Page tooltip shows the absolute path of the file.

[image]

luzpaz, 2021-08-05:
Closing as a dupe of #506

closed 2021-08-05 by luzpaz