realthunder/FreeCAD_assembly3 #511: Element count for Link produces wrong number of links

tomster, 2021-03-10:
Using 2020.0308 when I create a link from an existing body (wrapped in an assembly, so i can inherit its elements).
i then set the resulting link\s element count to 3. After hitting enter or tabbing away from the input field that number is then changed immediately (even before i could screen grab it) to 2:
[image]
and it produces two Links named ``Link_i0`` and ``Link_i2``
[image]
if i then stubbornly change it back to 3 it produces ``Link_i0``, ``Link_i2`` and another ``Link_i2`` where the latter two behave identically and cannot be placed individually:
[image]

in short, i cannot get the link array feature to work at all. any ideas?

tomster, 2021-03-10:
Oh and while I have your attention on the topic of links... :) I found that for creating an array whose count is linked to a spreadsheet value, it's easier to create the original part outside of the actual assembly and then create a link array with the actual number and then insert that into the assembly instead of adding the part to the assembly directly and then create an array with an element count of n-1. is that a good pattern? it would lead to creating a separate assembly which i would call "parts bin" where i create and collect all of the different parts that my product will consist of. there i can create and manage their elements and their names and then in the actual assembly i would only operate with link arrays where i set the element count to 1 or more.
does that make sense, is that a feasible approach?

realthunder, 2021-03-11:
I can't reproduce the problem. Can you please send me your `~/.FreeCAD/link.user.cfg` file? Are you doing this while in asm3 workbench? Are there any other add-on workbench you installed?

Regarding your link usage pattern. Yes, adding parts as links has many benefits. And it is also recommended to wrap any non-trivial part inside its own assembly, so that you can manually define its constraining elements, and possibly create many [interchangeable](https://github.com/realthunder/FreeCAD_assembly3/wiki/Replacing-Part) parts this way.

Another tip, you can easily replace any already added part with a link. Right click the part object in the tree view, select `Link actions -> Replace with a link`.

tomster, 2021-03-11:
indeed, removing `link.user.cfg` allows the feature to work again. i have never changed this file directly. i'm uploading it because it would be interesting to find out what caused this behaviour. i'm also uploading a diff to the newly created one, for convenience. looking at that i found an obvious candidate, namely the autocompute option, but switching that back on doesn't break anything.
and thanks for confirming the link pattern. and yes, trying out what "Replace with a link" does gave me the initial idea :)
[link.user.cfg.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6121361/link.user.cfg.zip)

tomster, 2021-03-11:
and yes, asm3 was active at the time. the only other workbench that is installed is a2+ and cadquery but neither were loaded.

tomster, 2021-03-11:
also, i'm wondering how can i influence the placement of the resulting links? in this case, for instance, the three items were scattered along the x and y axis seemingly at random and two of them occupying the same coordinates
[image]
my intuition was to set the axis placement of the link before setting the element count, but the above screenshot was created with these settings:
[image]
i have read the wiki page regarding links (https://github.com/realthunder/FreeCAD_assembly3/wiki/Link#app-namespace) and i can successfully manipulate the links' position *afterwards* using the documented properties:
[image]
what i' wondering about is how to influence it *before* creating the array.
this is because i would like to keep the count as a parameter and achieve a predictable (or better yet influencable) placement of the links when changing that number.

realthunder, 2021-03-12:
I still can't reproduce the problem with your config file. Can you reproduce it by restore your old config file?

Regarding the link element placement, if you want pattern you can try Draft array. By default, it internally uses Link as well. But note that since the array enforces a pattern, You'll get strange result if you try to constrain each individual element inside an assembly. You enable/disable draft array pattern by changing its `ExpandArray` property. Once the individual element is expanded, you move the elements at will.

tomster, 2021-03-13:
no icannot reproduce it with the old version of the config file. maybe having FC recreate it after moving it away fixed the issue as a side-effect. at this point i'm just going to close the issue and re-visit it, if it should come up again. thanks for trying to reproduce it, though!

closed 2021-03-13 by tomster