realthunder/FreeCAD_assembly3 #512: Sketch not added to active body when feature latest feature is hidden.

HaasNL2, 2021-03-11:
When I hide the latest body then try to add a sketch it places the sketch in the top, not the body.
Then when I try to move it back to the body, it sometimes work sometimes it throws an error to me.
Maybe I am doing something wrong?
Below recording:
[image]

realthunder, 2021-03-12:
You didn't capture the whole screen, so I am not sure, but are you creating sketch using the button in Sketcher workbench? The behavior there is that if you have a selection, then you are trying to map a sketch on it. So in your case, it is interpreted that you want to create a sketch that is attached to the body, which is a valid action. And the sketch must be put outside of the body to avoid cyclic dependency, i.e. a children cannot reference its parent.

If you create sketch by clicking the button inside PartDesign workbench, then it will ignore your body selection, and create the sketch inside the active body.

HaasNL2, 2021-03-12:
@realthunder Ok I think it is a bug at least it is not innovative, I checked which button I press. And the result of pressing the sketch button in the sketcher workbench has different results as the sketch button in the Part design workbench! The part design workbench places it in the body, and the Sketcher workbench places it on top of the body.

zardozerr, 2021-03-14:
I thought this was intended behavior of the sketch button in the sketcher wb vs the sketch button in part design. The button in the sketcher wb is more generalized, and doesn't care about bodies. You might be using it with the part wb, which doesn't use bodies, for example.

HaasNL2, 2021-03-14:
> 
> 
> I thought this was intended behavior of the sketch button in the sketcher wb vs the sketch button in part design. The button in the sketcher wb is more generalized, and doesn't care about bodies. You might be using it with the part wb, which doesn't use bodies, for example.

If this is so, I can work with this. But lets change the button name and Icon slightly because there is no way to know this otherwise. Maybe call it "Sketch to body" and just make the icon a bit more gray or so.