realthunder/FreeCAD_assembly3 #513: FreeCAD crash when compiled against realthunder/coin

ceremcem, 2021-03-13:
I'm trying to compile LinkStage3 against your fork of coin and I've run into several troubles. I removed `libcoin-dev` and `libcoin80c`, then I compiled and installed your fork of coin: 

```bash
#!/bin/bash
set -ue

cd $HOME
[[ -d coin ]] \
    || git clone --recurse-submodules https://github.com/realthunder/coin coin \
    && (cd coin; git pull)
cmake \
    -Hcoin \
    -Bcoin_build \
    -G "Unix Makefiles" \
    -DCMAKE_INSTALL_PREFIX=/usr/local \
    -DCMAKE_BUILD_TYPE=Release \
    -DCOIN_BUILD_DOCUMENTATION=OFF

cd coin_build
make
sudo make install
```

After issuing `ldconfig`, FreeCAD finally launched, but I got the following error: 

```
<class 'ModuleNotFoundError'>: No module named 'pivy'
```

I decided to compile Pivy from scratch: https://github.com/coin3d/pivy however it complained that I had a low version of `cmake` (mine was 3.13.4) so I installed cmake by: 

```
apt remove cmake
pip install cmake --upgrade
```

Now cmake version is 3.18.4 but Pivy again refused to compile because I needed Swig4 (and I have Swig3). I gave up trying to install Swig (and Pivy) but I recompiled FreeCAD with CMake 3.18.4. It took a whole recompilation (regarding to the compilation duration) and now, it interestingly doesn't throw the "Pivy not found" error, it just launches. 

However when I try to open a new document, FreeCAD crashes with the following backtrace: 

````
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x37840) [0x7efed6e46840]
```

### Version 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27679 (Git)
Build type: Release
Branch: LinkStage3
Hash: b7209f7f70b1e1a4c833d3f6da68e39fde8df51a
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.1
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

Original issue is here: https://github.com/ceremcem/build-freecad-asm3/issues/14

ceremcem, 2021-03-13:
Oh, sorry, I just remember that I manually installed `python3-pivy` via `apt-get` after upgrading CMake. The vanishing "Pivy not found" error is not related with CMake upgrade.

luzpaz, 2022-06-30:
@ceremcem is this still an issue?