realthunder/FreeCAD_assembly3 #514: Link placement not updating

tomster, 2021-03-13:
When I create a link array and edit the individual resulting links, sometimes (not sure yet when and when not) the values are erroneously evaluated to 0 (and the link is not moved as a result) eventhough the formula is evaluated correctly inside the edit dialog
here you can see the value ``0`` next to the formula:
[image]
but when i edit the value it shows that it correctly is evaluated to 400:
[image]
using 2021.0311

tomster, 2021-03-13:
update: after restarting the application i could change the ``400`` to ``200`` and it worked immediately (after hitting enter). when i then changed the value back to the formula this was updated, as well
[image]
during all of this (and prio to the restart, as well) there weren't any errors or other log messages on the console.

realthunder, 2021-03-16:
Can you please attach the whole file here? Are these links inside another assembly, with some constraints?. If so, then there will be some race condition there, as the expression and the assembly solver trying to compete to set the placement of the link.

tomster, 2021-03-20:
@realthunder you are right. I had competing constraints in the assembly. my strategy is now to not manipulate the placement of the link array members but simply treat the link array feature as a "mulitiplication engine" and perform any placement using constraints. (which then in turn manipulate those placements, if i'm beginning to understand the underlying mechanics?)

closed 2021-03-20 by tomster