realthunder/FreeCAD_assembly3 #515: Pocket parameters - Type - Up to face does not work

dreetie, 2021-03-16:
How to recreate the problem:
in part design -> make body and simple sketch, for example a square.
[image]

Make a sketch on a face to use as pocket (circle)
Pocket sketch and chose type: Up to face and select bottom face. 

[image]
Appimage used: FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210311-glibc2.12-x86_64
OS: Ubuntu 20.04.2 LTS

Even with a face selected the message: no face selected appears. 
With the freecad daily build from the main branch I don't have this problem

realthunder, 2021-03-17:
I can't reproduce this problem. Can you post your file here? If you can record a screencast, that'll be better.

realthunder, 2021-03-17:
https://user-images.githubusercontent.com/1207888/111398555-abf97680-86fe-11eb-8433-7aa6f0d47e1e.mp4

dreetie, 2021-03-17:
I think I found how to reproduce the problem: it only occurs when "Show preview" is not selected.

https://user-images.githubusercontent.com/22996477/111473976-dc0b3e80-872b-11eb-8e06-91212453ee11.mp4

realthunder, 2021-03-17:
I see. It is not strictly a bug then. Because you are selecting a face on the editing feature, but it is expecting you to select a face from some previous feature. It would be more user friendly to auto trace back the history of the face to the previous feature.

oficinerobotica, 2021-06-06:
Independent of the ShowPreview state, now Pad up to face works as expected.
Tested with:
```
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```

closed 2021-06-06 by oficinerobotica