realthunder/FreeCAD_assembly3 #517: Freecad automatically switche to first tab/document

nadarbreicq, 2021-03-17:
Hi !

I'm stuck with an issue. Don't know if it's a bug or a feature, but every time I work on multiple document, when I want to make a new sketch or everything else, FreeCAD select the first tab/document and I loose the focus on my document.
Someone got the same issue here  : https://forum.freecadweb.org/viewtopic.php?t=55306 ... It seems to be a bug in the 0.18 version...

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.311.24301 +3373 (Git)
Build type: Release
Branch: LinkStage3
Hash: 91ca94db328bf6126c4a01547b6aa2202d876ebc
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: French/France (fr_FR)
```

realthunder, 2021-03-17:
This annoying problem keeps coming back and haunting me. It seems never appear on Linux, more often in Mac, and sometimes Windows. Can you please provide a screencast showing the problem? Also please attach here your user configuration file in `c:\Users\<your_username>\AppData\Roaming\FreeCAD\link.user.cfg`

nadarbreicq, 2021-03-17:
Yhea, I made some researchs and it seems to be an annoying bug since 0.14 (as far as I know... ). I will send you a screencast tomorrow, but you can find le configuration file here : 
[link.user.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6160158/link.user.zip)

nadarbreicq, 2021-03-18:
Ok so today, I didn't manage to reproduce the issue with exactly the same files and without changing anything... So strange... I'll keep trying today...

realthunder, 2021-03-18:
Thanks for the appreciation!

Regarding this problem, it is related to how Qt handles focus change in child widget of the QMdiArea, i.e. those document sub-windows. So, to catch the problem, please pay attention to anything you are running that can grab focus in some unusual way.

nadarbreicq, 2021-03-18:
Okay ! I will pay attention to that. My screencast soft is ready to record if anything happened :D

irmantus, 2021-03-25:
I'm having same issue on MacOS. Only way to work around is to have only one file open and close "Start page" tab. But there is also one more bug that you can close "Start page" tab with "X" button, you have to do right click->close.

Screencast: https://youtu.be/3yQZIi67y1w

luzpaz, 2021-08-04:
Is this still a problem?

luzpaz, 2021-08-04:
dupe of #407 ?

nadarbreicq, 2021-08-04:
Never reproduced on my computer. Got a similar issue with one of my student some month ago. But after changing his version for a more recent one, he also never reproduced the issue.

luzpaz, 2021-08-04:
@irmantus is what you're seeing the same as #407 ?

Is this ticket safe to close ?

irmantus, 2021-08-04:
I can still reproduce this issue on the latest release.

MarcoPSP, 2021-11-10:
I am having the same problem, it makes it very annoying to have multiple files opened at the same time.
No messages in “report view”.
Is there a way to modify this behaviour using the python console?

OS: macOS 12.0
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +3962 (Git)
Build type: Release
Branch: LinkStage3
Hash: c6cb39697eb439caf7e5bd0208ca3082a7820547
Python version: 3.9.6
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

realthunder, 2021-11-11:
Are you using the BigSur image or the conda one? Can you please post a screencast showing the steps to reproduce the problem? Also please post your application settings here, `Tools -> Edit parameters -> Export`.

MarcoPSP, 2021-11-11:
> Are you using the BigSur image or the conda one? Can you please post a screencast showing the steps to reproduce the problem? Also please post your application settings here, `Tools -> Edit parameters -> Export`.

Hi real thunder, Thank you for your help and for your great branche.

Here you can find a screencast: [https://youtu.be/dCPWVJTWOO4](https://youtu.be/dCPWVJTWOO4)

The behaviour was observed just before the recording, also while adding a part. this case was to show the bug so the files were not saved and had the same name, but it happens with any saved project.

I am using the BigSur image, i didn't try the conda one, I will download it.

Here are my [application settings](https://drive.google.com/file/d/1Dw9N6Hv0xmw39M9E-E31-VRXfeiwdqkA/view?usp=sharing )

MarcoPSP, 2021-11-11:
Just tested with conda, i got the same behavior...

meprakarsa, 2022-01-21:
I experienced the same (using MacOS Catalina), and my colleague too (Mac OS Big Sur).
The strange thing is... it just happened recently (since last week only).

It Happened to FreeCad Link Daily & Stable (RealThunder branch), also daily main FC 0.20.
Simple operation like toggle visibility on/off (pressing space bar) reproduced this issue consistently, same with @irmantus .

Another strange behaviour: the problem resolved in 0.20 main branch after I reset the preference.
But when I applied the same technic to the RealThunder branches, it didn't work.
I have trialled ticked-unticked any suspicious preference setting, but nothing solves yet.

OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +4188 (Git)
Build type: Release
Branch: **LinkDaily**
Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git)
Build type: Release
Branch: **LinkStage3**
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: C/Default (C)

**## FreeCAD 0.20: ##**
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.20.25065 (Git)
Build type: Release
Branch: master
Hash: 4de2ad46b2b9c78ac2721a0dbea7c19306261095
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: C/Default (C)

pureselfmade, 2022-01-24:
Hello, I have the exact same issue as @irmantus described in the screencast: https://youtu.be/3yQZIi67y1w

I use freecad for production and design of technical drawings in my small company so it is a really annoying bug that kills the workflow. Any help is very much appreciated and I would like to make a donation if that can be solved!
 No matter if I use the Official stable or dev. releases or Realthunder's stable or daily. It keeps coming back with all of them. 

The weird thing is that it is not always behaving the same. Sometimes it does also switch to the first document in the tree view when I use the space bar to toggle the visibility of a part or body of another open document.
I do remember working for hours with multiple documents open in the past and not experiencing this issue. I have completely removed freecad from my computer and deleted all related files in the library folder but the issue keeps coming back.

Mostly I work with this setup:

OS: macOS High Sierra (10.13)
Word size of FreeCAD: 64-bit
Version: 0.20.27078 (Git)
Build type: Release
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: C/Default (C)

gotchoices, 2022-11-08:
I have this issue in the latest build.

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.30868 (Git)
Build type: Release
Branch: master
Hash: b3372a52054fc34c243ac508ce67717b1476f2a9
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * sheetmetal 0.2.58
[/code]

To reproduce:
- Create two separate documents.  
- In each one, create a body and a sketch
- Choose to edit the sketch in the second document

Results:  Instead of seeing the sketch you just chose to edit, the screen focus will change to the first part.  You won't be viewing the edited sketch (in the second part).  If you manually select that tab, you will see editing mode on the sketch has been selected--you just lost focus of it.

This issue also discussed in the forum [here:](https://forum.freecadweb.org/viewtopic.php?style=3&p=493530)

luzpaz, 2022-11-08:
Can anyone else confirm ? @nadarbreicq etc.. ?

lagnat, 2022-11-08:
I've seen that as well

gotchoices, 2022-11-08:
Sorry, I was confused about the distinction between the realthunder fork and the mainstream FreeCAD distro.  I downloaded the latest realthunder version and this bug is NOT present--seems to have been fixed (at least in my case).  I will report this in the main FreeCAD repo (where the bug is still present).  Here's the version I have working with the bug apparently solved:

OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 2022.1021.26244 +5249 (Git)
Build type: Release
Branch: LinkDaily
Hash: d307b5c936403e8dca3f24b5c1e64bd16c88b28d
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: C/Default (C)

Sorry for the confusion!

gotchoices, 2022-11-09:
Master ticket [Here](https://github.com/FreeCAD/FreeCAD/issues/7745)