realthunder/FreeCAD_assembly3 #518: Mirror feature - cannot mirror with "Transform sub-feature" box checked

rola22, 2021-03-17:
Hi.
As always sorry for poor English :)

I am not able to mirror when "Transform sub-features" option is enabled.
I don't get any error in report view.

Steps to reproduce:
- create simple Pad (tested with rectangular shape, doesn't matter if fully constrained),
- create pocket on one of the Pad faces,
- create Mirror -> select operations you want to mirror -> select reference plane -> click ok button,

Nothing happens. Mirror operation appears in history tree but the mirror is not created itself.

Now if you try to edit mirror feature and you disable "Transform sub-features" options you will get mirror with
all previously features done.

Tested with:
- Realthunders Daily release 2021.03.11
  System: Windows 10 Pro 64bit, build 19041
- Realthunders Daily release 2021.03.11 Linux Appimage
  System: Linux Mint 19.3

realthunder, 2021-03-18:
Can you post the file that 'Nothing happens'? By default (i.e. with Transform sub-features enabled), it creates pattern only uses the added or subtracted shape in each previous feature. With that option disabled, it will use the entire shape.

rola22, 2021-03-18:
Sure, here is a file.

[mirror problem - test file.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6162333/mirror.problem.-.test.file.zip)

I know how this feature works but for me it doesn't work correctly :).

If i select operations i want to mirror and i click ok i get empty mirror
When i edit this mirror none of the operations are selected anymore.

realthunder, 2021-03-18:
Yes there is indeed a bug in remembering the selected feature to mirror when you select in the editing task panel (which is fixed by the above commit). Selecting in the tree view works, though. Another thing, if none of the feature is selected, by default behavior is to use the previous feature, i.e. Pocket in your case. What do you mean you get an empty mirror, as in no shape in 3D view? If so, can you please post a screencast for that?

rola22, 2021-03-18:
Yes, i don't get mirrored shape in 3D view

[image]

realthunder, 2021-03-18:
Well that explains it. Because if Sub-feature transform is enabled, it only transforms the added or subtracted shape in each feature. In case of the Pocket, the subtracted shapes are those holes (i.e. cylinders), and when only the holes are mirrored with Pocket Face3 as the reference, they land in the air, nothing to be subtracted from. I haven't got time to make the preview of the pattern feature work yet. Once the preview works like the rest of the feature, it'll be obvious why there is no mirrored shape.

rola22, 2021-03-18:
I know that if i select Pocket only i will get nothing but as you can see on my screencast i've selected both Pad and Pocket to mirror. I don't understand why Pad is not mirrored.

Here is another screencast, this time with two Pads:

[image]

realthunder, 2021-03-18:
Yes, that is where the bug is. Selection in the editing task panel is broken. I've fixed it already. Meanwhile, as a workaround, you can select in the tree view.

https://user-images.githubusercontent.com/1207888/111633891-eebb6b80-8830-11eb-81c2-a91c2c5513c4.mp4

rola22, 2021-03-18:
Ok so if you fixed it already I think that we can close this issue.

Thanks for your feedback. Now I'm going to look for new bugs :)

I wish I could learn coding faster so i can help with developing :)

closed 2021-03-19 by rola22