realthunder/FreeCAD_assembly3 #519: Modernize model rendering

longmathemagician, 2021-03-17:
Apologies if this isn't the place for enhancement requests, with the shadowed view rendering mode realthunder added I believe this is the most appropriate place for this discussion.

Targets:
- Support more modern GLSL shader versions, in my reflection mapping testing I could only get freecad to accept version 110-130 features (possibly a fundamental issue with coin rather than a simple parameter change somewhere?)
- Add reflection mapped shaders -- not an important graphical effect for cad but is nice to have, and is somewhat supported by coin.
- Add support for ambient occlusion (or other screen space shaders). Ambient occlusion makes a huge difference in improving depth perception on complex assemblies. I'm not sure if this can be done entirely in freecad or if it would require further patches to coin3d.

realthunder, 2021-03-17:
Do you have experience on render development? I actually plan to work on that, but got delayed by other stuff. I am not going to use Coin3D for shaders. As the internal low level structure of Coin3D is quit out dated. Its high level stuff, like scene graph construction, navigation, save, restore are still relevant, and has a good design IMO.

So my plan is to keep the scene graph, construct the rendering list myself (already done in one of my branch), and then render using some other low level graphics library, such as [https://github.com/DiligentGraphics/DiligentEngine]this[/url]. It uses HLSL shader across all three platform. Scroll down on their github page to see their capability. It even has API for realtime raytracing, imagining that in FreeCAD.

HaasNL2, 2021-03-18:
I started an issue on the [Blender cycles stand alone page ](https://devtalk.blender.org/t/building-cycles-standalone-not-working-for-me/17861/19)to get a windows build but apparently it already works in Linux and Mac.
If you could make it so we can just turn Cycles on and off it would be awesome. I do prefer cycle since we can use use material database that cycles have, and we can export it to blender and do animation and more professional renders.