realthunder/FreeCAD_assembly3 #520: Stable Branch

ceremcem, 2021-03-17:
We need a stable branch. We also need features. Features should be tested to make it "stable". If we use stable branch, who will test the new features? How will those new features become stable?

Yes, this is a complex problem. 

However, we need a stable "thing". When we demonstrate LinkStage3 to an important user/company, it shouldn't fail while performing most basic functions, like PartDesign. This happened today and this is the second big fish. 

Even AppImage's should contain both the stable and the testing/tip version and rollback to the stable one on demand.

(Related: #413)

realthunder, 2021-03-17:
I am actually planning for a more stable release. I guess the difference is to just not release new features right away, and only include fixes for each release. Only merge in new features when they are tested for some time in the daily release. The problem is, of course, once there are stable and daily release, users tend to download the stable release...

ceremcem, 2021-03-17:
Please help me become clear: Do you think the [Making projects resistant to breaking changes](https://forum.freecadweb.org/viewtopic.php?f=8&t=54298) proposal won't work as advertised or it's not the correct time to "support" it?

HaasNL2, 2021-03-18:
I'll keep testing unstable builds on windows. Don't mind a bit of crashes, if I can help make this build more awesome!
If we could use Chocolatey for windows builds that would be cool! I sue that currently to update Freecad with, saves me the hassle of downloading and unzipping. https://chocolatey.org/packages/freecad

Rabbit-sk, 2021-03-27:
[Stable 2021.03.26](https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11) 🚀

HaasNL2, 2021-03-27:
Damn nice work!!!

closed 2021-05-08 by ceremcem

ceremcem, 2021-06-14:
@realthunder Where is the stable branch under `https://github.com/realthunder/FreeCAD/tree`?

realthunder, 2021-06-14:
The `Stable` branch is the default `LinkStage3` branch, which is what's shown in that URL. The `Daily` is `LinkDaily`.

NothanUmber, 2021-07-18:
Considering using the daily vs stable vs 0.19 builds. One question: Are files saved in daily still compatible with stable? Or stable/daily still compatible with 0.19? When experiencing showstopper bugs, then (reporting and) going to a more stable version in the meantime should be ok for many noncritical works. If one has to either wait for a fix or restart from scratch with stable/0.19, then these incompatible versions would mostly be viable for dedicated testing sessions, less "testing by using".
So long story short:
* Are files saved with daily/stable currently still fully compatible with stable/0.19?
* If yes/no: Could it make sense to have a compatibility table (with at least Linkstage stable and the at least one official, most compatible FreeCad release) at the beginning of each release history?

realthunder, 2021-07-19:
FreeCAD is designed to be modular. Each module implement the logic behind its object following a common data structure. If you run a version of FreeCAD (regardless its upstream or my branch) with some module not installed the common `Shape` data of the object can usually be read and displayed, but you can no longer change the parameters of the object.

Here is the level of compatibility I am aiming for. My branch shall be able to read files saved in upstream, and you shall be able to change the parameters and modify your model. The upstream shall be able to read the file saved in my branch, but there is no guarantee for you to be able to modify the model. And if you do, the result may not be as expected. 

The same level of compatibility applies to my `Stable` vs. `Daily` release. And BTW, this also applies to old and new releases of official FreeCAD, e.g. 0.18 vs. 0.19.

NothanUmber, 2021-07-20:
Thanks @realthunder for the feedback. Hm, then for models where the parametric aspect is important it might be best to use upstream.
But it's good to know that a model from LSstable/daily can at least be loaded in upstream, so there is no need to e.g. archive step files (e.g. for the case that the LinkStage version with which a model was created doesn't run on a new OS anymore).

Hope one day the CAD world settles on a default export format that also saved the construction history. Of course it couldn't support all the bells and whistles of the individual files, but basic concepts how a model can be made parametric appear (from a layman's perspective) very similar in most history based CADs.
Or the format of a particular CAD gets licensed freely and can be imported and exported by all the others. But there politics might be (even more) difficult than the technical side perhaps :)
Is there e.g. a spec of the FreeCad file format so that Dassault or Siemens could support that as import/export format without having to use the actual GPL code from FreeCad?

NothanUmber, 2021-07-20:
The newest STEP iterations sound promising, they seem to support constraints and construction history to some point: http://www.ap242.org/edition-2
Doesn't help though that the specification for the standard is not publically available (at least I get access denied when following the link to the white paper)...

Rabbit-sk, 2021-07-20:
> Hm, then for models where the parametric aspect is important it might be best to use upstream.

It depends on which branch you consider better for your work. Since RealThunder's branches have many improvements that have not yet reached the upstream, it is logical that the upstream branch may not be able to change the objects created in the RealThunder's branches. I've been using FreeCAD from RealThunder's branches for a long time for serious work, and so far I have had no reason to use the upstream branch. For me, the branch is better, which contains improvements, thanks to which the work can be done faster and changes to older modeling steps can be made more flexible, without breaking the model. 

https://user-images.githubusercontent.com/74265320/126364238-b7360cba-0e33-489b-aa23-ab26e6b8bade.mp4

https://user-images.githubusercontent.com/74265320/126365401-de3b93f9-e4b9-476e-849c-1eb831c08836.mp4

NothanUmber, 2021-07-20:
Makes sense, thanks for the demo, neat, indeed! :)

ceremcem, 2021-09-10:
@realthunder I'm when do you plan to merge the latest "Configurable objects" behavior to the stable branch?

realthunder, 2021-09-10:
It's been merged already.

luzpaz, 2022-04-08:
@realthunder can this ticket be unpinned ?