realthunder/FreeCAD_assembly3 #524: Toolbits missing parameters

twardnw, 2021-03-20:
```
OS: macOS Mojave (10.14)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.311.24301 +3373 (Git)
Build type: Release
Branch: LinkStage3
Hash: 91ca94db328bf6126c4a01547b6aa2202d876ebc
Python version: 3.7.10
Qt version: 5.12.1
Coin version: 4.0.1
OCC version: 7.4.0
Locale: C/Default (C)
```

Trying to work on some CAM paths, loading the ToolBit Library editor, I followed the dialog to set up the Library directory, and allowed FreeCAD to import the defaults. However, none of the bits have parameters aside from shape? And attempting to recompute the path using any of them leads to errors like:

```
PathToolBit.ERROR: Did not find a PropertyBag in v_bit - not a ToolBit shape?
```

```
10:29:13  Running the Python command 'Path_Engrave' failed:
Traceback (most recent call last):
  File "/Applications/FreeCAD-asm3.app/Contents/Resources/Mod/Path/PathScripts/PathOpGui.py", line 1319, in Activated
    return Create(self.res)
  File "/Applications/FreeCAD-asm3.app/Contents/Resources/Mod/Path/PathScripts/PathOpGui.py", line 1283, in Create
    obj = res.objFactory(res.name)
  File "/Applications/FreeCAD-asm3.app/Contents/Resources/Mod/Path/PathScripts/PathEngrave.py", line 151, in Create
    obj.Proxy = ObjectEngrave(obj, name)
  File "/Applications/FreeCAD-asm3.app/Contents/Resources/Mod/Path/PathScripts/PathEngrave.py", line 52, in __init__
    super(ObjectEngrave, self).__init__(obj, name)
  File "/Applications/FreeCAD-asm3.app/Contents/Resources/Mod/Path/PathScripts/PathOp.py", line 193, in __init__
    job = self.setDefaultValues(obj)
  File "/Applications/FreeCAD-asm3.app/Contents/Resources/Mod/Path/PathScripts/PathOp.py", line 338, in setDefaultValues
    obj.OpToolDiameter = obj.ToolController.Tool.Diameter

'FeaturePython' object has no attribute 'Diameter'
```

I also experienced near the same behavior on the same Linux release.

luzpaz, 2021-08-04:
@realthunder quick question, is this at all related to incomplete yet pending upstream development?

johnsonm, 2021-09-26:
I see that commit 3524995a7 is referenced as in Daily, but I got the same message running FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210922-glibc2.12-x86_64.AppImage

```
14:05:21  PathToolBit.ERROR: Did not find a PropertyBag in endmill - not a ToolBit shape?
14:05:21  PathUtil.WARNING: 2.5 mm Endmill has no property CuttingEdgeHeight (CuttingEdgeHeight))
14:05:21  PathUtil.WARNING: 2.5 mm Endmill has no property Diameter (Diameter))
14:05:21  PathUtil.WARNING: 2.5 mm Endmill has no property Length (Length))
14:05:21  PathUtil.WARNING: 2.5 mm Endmill has no property ShankDiameter (ShankDiameter))
14:05:27  <PropShape> PropertyTopoShape.cpp(412): Pending recompute for generating element map: endmill#Body
14:05:27  <PropShape> PropertyTopoShape.cpp(412): Pending recompute for generating element map: endmill#Sketch
14:05:27  <PropShape> PropertyTopoShape.cpp(412): Pending recompute for generating element map: endmill#Revolution
14:05:27  <PropShape> PropertyTopoShape.cpp(412): Pending recompute for generating element map: endmill#Revolution
```

```
{
  "version": 2,
  "name": "2.5 mm Endmill",
  "shape": "endmill.fcstd",
  "parameter": {
    "CuttingEdgeHeight": "17.000 mm",
    "Diameter": "2.5 mm",
    "Length": "38.000 mm",
    "ShankDiameter": "3.175 mm"
  },
  "attribute": {
    "Flutes": "2"
  }
}
```

I have confirmed that the parameters are in endmill.fcstd

luzpaz, 2022-07-11:
Is this still an issue?

nik570, 2023-06-09:
Yes it still an issue. I have to computers and same problem on one of them but on the this problem does not exist. Use Debian and tried both 0.19 installed using package manager and appimage 0.20 on both computers. It is related to computer but version is not important.

Tried to delete ./Freecad foldet did not make a difference but guess there is some value stored somewhere.