realthunder/FreeCAD_assembly3 #528: Spacemouse default settings different between vanilla FreeCAD and asm3 builds

julia-m-stephenson, 2021-03-22:
Hi,

I've just noticed a difference using my Space Mouse between the last (stable?)  Daily build for Windows 10 and the current vanilla 0.19 release. 

I was just about to submit the following issue when I realised that the Spacemouse default settings were wildly different between the standard executable and the assembly3 daily.  I viewed the Space mouse settings from within the "3D Connexion Home" App.

This is what it should look like.

[image]

This may be because the executable has a different name? Does FreeCAD pass a configuration to the Spacemouse driver? 

I guess this may cause other Spacemice users a problem too.

Best Regards and thanks for your awesome work.
Julia

Ignore the following - this is symptoms this was causing....

With the standard release I can pitch, roll, rotate zoom in and out nicely as expected. 

With the 11th March Daily build I'm finding rotating round the X-axis (Front to Top to Rear to Bottom) is jerky and the model moves either towards or away from you at the same time. Rotating round the other axis (Y,Z) works smoothly as does zoom in / out.

To test this I have a simple model basically a rectangle with two corners removed and arch shaped pocket (I was trying out the Topilogical issue from this Youtube video - https://www.youtube.com/watch?v=0UlnrG-2atk)

Let me know if you need any other details.
Best Regards and many thanks for your awesome efforts
Julia

Windows Version: Windows 10 Pro, V1909, OS Build 18363.1440
Space mouse driver is 1.4.0.12688

Daily:
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.311.24301 +3373 (Git)
Build type: Release
Branch: LinkStage3
Hash: 91ca94db328bf6126c4a01547b6aa2202d876ebc
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)

Vanilla FreeCAD:
OS: Windows 10 Version 1909
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)

julia-m-stephenson, 2021-03-22:
This is what the asm3 daily brings up as defaults

[image]

realthunder, 2021-03-24:
I just tried it on my laptop. And I used the space mouse for the first time with upstream FreeCAD. It has the same default settings as your second picture. I guess the settings are stored per executable. So after you have manually change the settings, does it working fine now, in my branch?

julia-m-stephenson, 2021-03-24:
Hi, 

Yes it is working fine in your branch once the settings are changed. I was in two minds whether to submit this, as its more of a configuration/documentation issue but it does detract from the out-of-the-box user experience.

I installed 0.19 upstream and that came up like the first picture, maybe it picked up the settings from my 0.18 install? I've played around with the various 3dConnexion SDK but was concentrating on reading input rather than the driver config side.

Thanks for taking a look - I can't use 3D software without the space mouse. I think of it as you pick up the object with your left hand using the Spacemouse and work on it using your right hand. Its very intuitive....

Best Regards
Julia

herrgerd, 2021-03-28:
I'm using CAD software for like 10 years now and rotating around a specific point is something FreeCAD still can't do.
I don't know why, but after using SolidWorks, Inventor or SolidEdge (which I use to earn my daily monies), the movement and rotation in FreeCAD feels totally awkward, regardless of any setting I have tried.

I hoped the SpaceMouse would make a difference here, but it doesn't. I'll try to do a short screencast of the same part in SolidWorks and FreeCAD.

I'll zoom far out and try to get the part aligned on the screen with just the SpaceNavigator.

julia-m-stephenson, 2021-03-29:
I'm going to close this issue. Jan you should create a new ticket as your issue is different.

closed 2021-03-29 by julia-m-stephenson