realthunder/FreeCAD_assembly3 #530: Not retro-compatible with 0.19 Daily

pierbout, 2021-03-23:
If I open in Real Thunder a file created with "0.19 daily", make a modification, save the file under Real Thunder and then reopen it in 0.19, most objects are called "Unnamed" in the "Model" window and I loose the tree (all objects are at the same level.

Maybe a problem of retro-compatibility of the XML file?

Messages:
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  XML Attribute: "Ref" not found
11:21:55  Invalid Document.xml: End of document reached
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Enumeration index -1 is out of range, ignore it
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Enumeration index -1 is out of range, ignore it
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Enumeration index -1 is out of range, ignore it
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Enumeration index -1 is out of range, ignore it
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:21:55  SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
11:21:55  SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
11:21:55  SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  XML Attribute: "Ref" not found
11:27:22  Invalid Document.xml: End of document reached
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Enumeration index -1 is out of range, ignore it
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Enumeration index -1 is out of range, ignore it
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Enumeration index -1 is out of range, ignore it
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Enumeration index -1 is out of range, ignore it
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  Extension is not a python addable version: 'PartGui::ViewProviderAttachExtension'
11:27:22  SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
11:27:22  SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
11:27:22  SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices

Note: When I open the 0.19 file in real thunder, it says it must recompute. Same result when I say yes or no (if I say yes, the recompute partially fails)

[image]

realthunder, 2021-03-24:
You 0.19 download is too old. The `Sketcher` workbench in upstream FreeCAD has a major internal changes recently. You won't be able to restore a sketch saved in the latest upstream either.

pierbout, 2021-03-24:
I updated the daily, and guess what... it works! Thanks a lot!

closed 2021-06-06 by oficinerobotica