realthunder/FreeCAD_assembly3 #531: Cannot export a file created in 0.19 Daily

pierbout, 2021-03-23:
Hello,

If I open my project, created in Freecad 0.19 and try to export it to STL, I get no error message but no STL file . 

Note:
- The export works fine in Freecad 0.19
- When I open the 0.19 file in real thunder, it says it must recompute. Same result when I say yes or no (if I say yes, the recompute partially fails)
- This is my first install of real thunder. So I don't know if this issue existed previously or not.

**Linux Mint 20
Real Thunder 2021/03/11
Freecad 0.19 revision 22665**

realthunder, 2021-03-24:
Can you please attach here your FCStd file created in 0.19?

pierbout, 2021-03-24:
Hello,
Here is the file . Thanks for your help.
[projet-CT6H.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6196158/projet-CT6H.zip)

oficinerobotica, 2021-06-06:
Tested with 
```
OS: KDE neon User Edition 5.21 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```
and the issue isn't present anymore.

closed 2021-06-06 by oficinerobotica