realthunder/FreeCAD_assembly3 #534: Improvement proposal: Focus on tree view when adding a new Part container

Rabbit-sk, 2021-03-24:
Dear Realthunder,

When a new Part container is created, it is highlighted and becomes active, but if I want to rename it with F2 right after creation, this is not possible. Can it be "clicked" after creation, as in the case of a [new Body](https://github.com/realthunder/FreeCAD_assembly3/issues/416)?

[image]

Best regards

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.311.24301 +3373 (Git)
Build type: Release
Branch: LinkStage3
Hash: 91ca94db328bf6126c4a01547b6aa2202d876ebc
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: Slovak/Slovakia (sk_SK)

tomster, 2021-03-24:
I second this. I've learned that giving everything meaningful names right from the beginning makes the whole experience later on during the assembly process *a lot* more pleasant, so I think this little change might go a long way...

Rabbit-sk, 2021-03-25:
> Default shortcut 'Shift + F2', for easy renaming of newly created object

By default, can this shortcut be set to the `F2` key, or can the user later change it to the `F2` key? It is impractical to press the `Shift + F2` key combination with one hand, and it is also impractical to leave the mouse to press it with two hands.

realthunder, 2021-03-25:
Maybe not. F2 is a built in shortcut used by the tree view. But you can assign a different shortcut for this `Std_RenameActiveObject` command. It's in the `Edit` menu.

Rabbit-sk, 2021-03-25:
> But you can assign a different shortcut for this Std_RenameActiveObject command.

Okay then. Thank you!

closed 2021-03-29 by Rabbit-sk