realthunder/FreeCAD_assembly3 #536: Path workbench does no longer work

snelweg, 2021-03-24:
Very strange things are happening with the Path WB in the linkstage3 version
Tool editor is not working, paths and operations do no work (can  not even be setup)

Problems were  noticed today on my Windows machine in my workshop, I also installed the latest linkstage3 version there, but still nothing worked.  At home I use the latest linkstage3 version for Linux, nothing from the Path WB worked properly.

I switched to the 0.19.1 stable version and it all worked again.

Asm3 version

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.311.24301 +3373 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 91ca94db328bf6126c4a01547b6aa2202d876ebc
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

stable version
> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.19.1)
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

snelweg, 2021-03-24:
Errors/warnings in Linkstage3 version, and nothing works. 

> 19:07:11  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: endmill#Body
19:07:11  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: endmill#Sketch
19:07:11  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: endmill#Revolution
19:07:11  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: endmill#Revolution
19:07:17  <Gui> ViewProviderDocumentObject.cpp(779): no view
19:07:17  <Gui> ViewProviderDocumentObject.cpp(779): no view
19:07:17  <App> PathToolBit.py(304)|GeoFeatureGroupExtension.cpp(268): Remove invalid member Unnamed#PropertyBag from Unnamed#Body001
19:07:17  Auto correct group member for Unnamed#Body001
19:07:17  PathToolBit.ERROR: Did not find a PropertyBag in endmill - not a ToolBit shape?
19:07:26  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: endmill#Body
19:07:26  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: endmill#Sketch
19:07:26  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: endmill#Revolution
19:07:26  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: endmill#Revolution
19:07:27  <Gui> ViewProviderDocumentObject.cpp(779): no view
19:07:27  <Gui> ViewProviderDocumentObject.cpp(779): no view
19:07:27  <App> PathToolBit.py(304)|GeoFeatureGroupExtension.cpp(268): Remove invalid member Unnamed#PropertyBag001 from Unnamed#Body001
19:07:27  Auto correct group member for Unnamed#Body001
19:07:27  PathToolBit.ERROR: Did not find a PropertyBag in endmill - not a ToolBit shape?
19:07:27  PathUtil.WARNING: 5mm Endmill has no property CuttingEdgeHeight (CuttingEdgeHeight))
19:07:27  PathUtil.WARNING: 5mm Endmill has no property Diameter (Diameter))
19:07:27  PathUtil.WARNING: 5mm Endmill has no property Length (Length))
19:07:27  PathUtil.WARNING: 5mm Endmill has no property ShankDiameter (ShankDiameter))

snelweg, 2021-03-24:
Linkstage3 problems:

https://user-images.githubusercontent.com/1257901/112362828-26b12b80-8cd5-11eb-86fd-6309dd874e25.mp4

Current FC stable version (expected behaviour):

https://user-images.githubusercontent.com/1257901/112362935-45172700-8cd5-11eb-946c-3015f0d14728.mp4

snelweg, 2021-03-25:
I hope there is a fix soon, I miss all the goodies that Linkstage3 brought. Looks like I am back in time with the official stable release.

(there are some errors in the stable release too, the tools in the tooltable are not correctly setup. I guess one would only notice that with a fresh install).

luzpaz, 2021-03-25:
> (there are some errors in the stable release too, the tools in the tooltable are not correctly setup. I guess one would only notice that with a fresh install).

//cc @sliptonic

realthunder, 2021-03-25:
This is the same problem reported in #524, and is fixed by the commit referenced there. I'll release a new version soon.

snelweg, 2021-03-25:
> > (there are some errors in the stable release too, the tools in the tooltable are not correctly setup. I guess one would only notice that with a fresh install).
> 
> //cc @sliptonic

I tried to file an issue at https://tracker.freecadweb.org/my_view_page.php but I soon left, what an ancient system and so user unfriendly.
I will file the issue here and include @sliptonic

closed 2021-03-25 by snelweg