realthunder/FreeCAD_assembly3 #538: BIM Door creation issue

QuakeString, 2021-03-26:
Unable to substract door from a wall
[https://youtu.be/zF187cZFP7E](https://youtu.be/zF187cZFP7E)
above link to the issue.
Appimage version 2021.311

my system Manjaro (21.0) KDE Plasma 5.21.3

realthunder, 2021-03-26:
Can you please post a sample file showing the problem?

QuakeString, 2021-03-26:
> Can you please post a sample file showing the problem?

[here is the FCStd file](https://drive.google.com/file/d/1H9-RdJ2pzraKYTvdBT3ndDw-Jc2DbRcP/view?usp=sharing)

realthunder, 2021-03-27:
There is some deeper problem in the Arch Window and Wall object here. The object dependency is inverted for some reason (i.e. the Window object depends on the Wall, but by right it should the parent object depends on the child). This cause the objects to be recomputed in wrong orders. If you open your attached file in the upstream FreeCAD, and mark the whole document for recompute, you'll also get some incorrect behavior. The door will jump around each time you mark the recompute.

With my branch the first time recompute has no effect, because of the wrong dependency order, i.e. the Wall is recomputed before the door. Should be the other way round. You'll have to manually mark the Wall for recompute to get the result. I'll raise this issue in the forum later.

QuakeString, 2021-03-27:
Thanks for explaining, i did try to recompute, and it does worked.

luzpaz, 2021-06-29:
CC @yorikvanhavre