realthunder/FreeCAD_assembly3 #545: New renderer -> model disappears

ngocminhdao88, 2021-03-29:
When trying new renderer, my model disappears. Switching back to auto, distributed or centralized shows the model on the screen again.

Here is my model, created from older version of Linkbranch:
[computer-table.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6223009/computer-table.zip)

Here is my display settings:
[image]

My FreeCAD info:
OS: Ubuntu 20.04.2 LTS (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.22506 +5276 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ff1590e1d7c0aeaf45a29941ec20a10d54cf92ff
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

oficinerobotica, 2021-03-30:
I can confirm. For me it happens randomly only at first opening of some models , closing and reopening the file whitbought restarting FC usually solves the problem.  If I click somewhere in the empty 3d view things get selected in the tree so the model is there but just lacks it's visual representation.

oficinerobotica, 2021-06-03:
This issue has been fixed by commit  https://github.com/realthunder/FreeCAD/commit/d508f2f80b1671bdda664bc4cc6dc811363df070

closed 2021-06-03 by oficinerobotica