realthunder/FreeCAD_assembly3 #549: FreeCAD freezes when using subshapebinder in new daily version

ngocminhdao88, 2021-03-30:
FreeCAD freezes when i try to use subshapebinder function in PartDesign Workbench (**daily** version).

To reproduce the issue:
PartDesign Workbench -> Create a sketch (circle for example) in a body 1 -> Create a new body 2 -> Copy the geomeotry from body 1 using Subshapebinder into body 2 >> FreeCAD freezes.

The stable version works fine.
> OS: Ubuntu 20.04.2 LTS (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.24301 +3441 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 0a157a307c91d241a370cdda6d8f0471b7c48915
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

My daily version:
> OS: Ubuntu 20.04.2 LTS (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.22506 +5276 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ff1590e1d7c0aeaf45a29941ec20a10d54cf92ff
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

oficinerobotica, 2021-03-31:
I don't know if this is related , if not I will open a new issue.
In the attached file there are shapebinders belonging to <Master_Case> body  that reference the <RaspberryPi2 STEP214> part. 
If the position of the <RaspberryPi2 STEP214> is changed the recompute will go on forever. The system monitor reports that the application is processing, not actually freesed but after 10 mins of waiting I killed it .

[Rasberry_Case.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6238492/Rasberry_Case.zip)

papamat, 2021-04-01:
I can also confirm this. 

- New File 
- Part Design
- Crete a sketch
- Draw an arbitrary shape and confirm
- Select sketch and press the green shape binder button
Freecad freezes

Killing freecad with "-SIGSEGV" gives the following output

`Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3da60) [0x7f36a7b58a60]
#1  0x7f3695f79fc3 in Part::TopoShape::makEWires(Part::TopoShape const&, char const*, bool, double) from /tmp/.mount_FreeCAj6kRVL/usr/lib64/Part.so+0x3
#2  0x7f3695dc283e in Part::SubShapeBinder::update(Part::SubShapeBinder::UpdateOption) from /tmp/.mount_FreeCAj6kRVL/usr/lib64/Part.so+0x320e
#3  0x7f3695dc82d7 in Part::SubShapeBinder::onChanged(App::Property const*) from /tmp/.mount_FreeCAj6kRVL/usr/lib64/Part.so+0xad7
#4  0x7f36a9277ae5 in App::Property::touch() from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib64/libFreeCADApp.so+0x125
#5  0x7f36a9274aa8 in App::Property::hasSetValue() from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib64/libFreeCADApp.so+0x68
#6  0x7f36a92d2ae2 in App::PropertyXLinkSubList::setValues(std::map<App::DocumentObject*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::less<App::DocumentObject*>, std::allocator<std::pair<App::DocumentObject* const, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > > >&&) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib64/libFreeCADApp.so+0x512
#7  0x7f3695dc436b in Part::SubShapeBinder::setLinks(std::map<App::DocumentObject*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::less<App::DocumentObject*>, std::allocator<std::pair<App::DocumentObject* const, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > > >&&, bool) from /tmp/.mount_FreeCAj6kRVL/usr/lib64/Part.so+0xeeb
#8  0x7f35d3cbbabc in PartGui::makeSubShapeBinder(Base::Type) from /tmp/.mount_FreeCAj6kRVL/usr/lib64/./PartGui.so+0xd6c
#9  0x7f360094b4cc in CmdPartDesignSubShapeBinder::activated(int) from /tmp/.mount_FreeCAj6kRVL/usr/lib64/PartDesignGui.so+0x2c
#10  0x7f36a9dbd4c7 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib64/libFreeCADGui.so+0x3e7
#11  0x7f36a8067a18 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Core.so.5+0x6c0
#12  0x7f36a8874928 in QAction::triggered(bool) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0x32
#13  0x7f36a88762b2 in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0xc6
#14  /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5(+0x20dc3c) [0x7f36a8912c3c]
#15  0x7f36a8912d19 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0x79
#16  0x7f36a89a941c in QToolButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0xa
#17  0x7f36a88a44ed in QWidget::event(QEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0x133
#18  0x7f36a89a94a9 in QToolButton::event(QEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0x45
#19  0x7f36a88797f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#20  0x7f36a887df5e in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0x70a
#21  0x7f36a9d6cb39 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib64/libFreeCADGui.so+0x89
#22  0x7f36a8051606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Core.so.5+0x86
#23  0x7f36a887d4c4 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0x286
#24  /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5(+0x1aefd4) [0x7f36a88b3fd4]
#25  /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5(+0x1aff7c) [0x7f36a88b4f7c]
#26  0x7f36a88797f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#27  0x7f36a887f31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#28  0x7f36a9d6cb39 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib64/libFreeCADGui.so+0x89
#29  0x7f36a8051606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Core.so.5+0x86
#30  0x7f36a83f5ec5 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAj6kRVL/usr/bin/../lib/libQt5Gui.so.5+0x677
[... cut ...]
#42  /lib64/libc.so.6(__libc_start_main+0xf2) [0x7f36a7b431e2]
#43  /tmp/.mount_FreeCAj6kRVL/usr/bin/FreeCADLink(+0x47d5) [0x560468a4f7d5]
`

The Freecad version I am running is the following
> OS: Fedora 33 (Workstation Edition) (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.22506 +5276 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ff1590e1d7c0aeaf45a29941ec20a10d54cf92ff
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

oficinerobotica, 2021-06-03:
I'm not really sure where the commit that fixed this was but I don't experience this problem anymore. Should this issue be closed?

papamat, 2021-06-03:
I can also confirm that is fixed.

closed 2021-06-03 by oficinerobotica