realthunder/FreeCAD_assembly3 #551: PartDesign pad up to face won't register face selection .

oficinerobotica, 2021-03-31:
PartDesign pad up to face won't register face selection in the latest daily. Sorry for the dry report but it is just that. Selection of a face during a pad task with "up to face" option won't transfer the face ID in to the "Face" field thus yelding the error "no face selected"

BR

realthunder, 2021-03-31:
Have you turned off `Show preview`? Because if so, then the object shown is the Pad itself, while the UpToFace property expects face from some previous feature. I'll later add some code to trace the Pad's select face to its base object.

oficinerobotica, 2021-03-31:
Yes, the `Show preview` was off, turning it back on allowed the face selection. Thank you. I'll close the issue.

closed 2021-03-31 by oficinerobotica