realthunder/FreeCAD_assembly3 #554: .amf export does not work with folders containing Umlauts (äüö)

herrgerd, 2021-04-02:
Hi there.
Two things are strange and only happen when using one of the **Umlaut-characters like ä, ü, ö.**
With .stl and other formats nothing of the following happens!

For .amf exports:
1. if the folder contains one of them, exporting your part into that folder does not work at all
2. if your file contains one of those characters, the filename is mangled

Saving as a file with an Ä:
[image]

Resulting filename:
[image]

Resulting filename when saving as a file with an Ö:
[image]

Cheers, Jan

luzpaz, 2022-01-12:
@herrgerd please retest on latest Daily and report back. Thanks!

herrgerd, 2022-01-12:
> @herrgerd please retest on latest Daily and report back. Thanks!
.amf and .stl export do work for me in the latest daily.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +3962 (Git)
Build type: Release
Branch: LinkStage3
Hash: c6cb39697eb439caf7e5bd0208ca3082a7820547
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

luzpaz, 2022-01-12:
@herrgerd you aren't using latest Daily actually. please update and retest. thanks

herrgerd, 2022-01-13:
> @herrgerd you aren't using latest Daily actually. please update and retest. thanks
Sorry, I have multiple versions (to fall back to a earlier one if bugs and stuff) and started the wrong one.

Anyway, still works:
[image]

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: German/Germany (de_DE)