realthunder/FreeCAD_assembly3 #556: Spreadsheet and configuration table bugs and sugestions

MarcelP102, 2021-04-04:
Dear Realthunder,

I've experiencing some problems using spreadsheet and the configuration table. When setting up parametric objects you often want to add or reorder the spreadsheet items. Right know this is not possible:
- Inserting a row above configuration table gives an error (<Exception> PropertyStandard.cpp(586): PropertyEnumeration Test_2#Box.Test expects type in list to be string, not tuple) I have also tried to start the configuration table on the third row. This gives the same error when adding a row above first row.
- Inserting a row within the options of the configuration table gives the same error. Inserting between C en D for example (row 7 and 8)
- After setting up the configuration table I can't add more columns, for example Width. (error: <Spreadsheet> SpreadsheetDelegate.cpp(66): Bound cell C5 cannot be edited) The solution I found is to 'unsetup' the configuration table, make the new options with alias and then remake the configuration table.  This is not really user friendly.
- After creating the configuration table I can't rename the group. In the example I have the group 'testing' I can't change this name. After opening the 'setup Configuration Table' The Group text is empty? Entering another name won't change the name in the Property View.

Feature request:
- Would be nice If there was an option to reorder the rows and column (move up/down left/right)
- Would appreciate if you could write a little bit about how to use the cells 'edit mode'. Especially the Button, Combobox, Label and Quantify aren't clear to me.

Example Project: [Test 2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6254122/Test.2.zip)

Thanks for considering this.
Best Regards, Marcel

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.24301 +3441 (Git)
Build type: Release
Branch: LinkStage3
Hash: 0a157a307c91d241a370cdda6d8f0471b7c48915
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: Dutch/Netherlands (nl_NL)

herrgerd, 2021-04-05:
Oh @MarcelP102 you're addressing a few very valid points here. 
I stopped using the spreadsheet function because of this a long time ago. Moving cells always means you have to "re-link" those: annoying. To be fair, the official docu says so under "Limitations": 

"The Cell Properties described above are not copied."

Regarding the "edit modes": I searched for like an hour in two languages and only found scraps of information here and there. The FreeCAD documentation is very basic on the workbench, too:

https://wiki.freecadweb.org/Spreadsheet_Workbench
https://wiki.freecadweb.org/Manual:Using_spreadsheets

realthunder, 2021-04-05:
The `cannot edit group` is a bug. I'll fix that. You can manually rename the group name of any dynamic property in the property view, by enable `Show all`, and select `Rename property group`. The row insertion above configuration table is not supported at the moment. I probably should have disabled explicitly. Inserting rows in-between the table should be working fine. I've tested it. Maybe you can make a screencast showing the error? To insert rows before the table, please click `unsetup` button when you bring up the configuration table. And make the table again after adjustment.

[image]

The `Edit mode` feature only exists in my branch. And it is documented [here](https://github.com/realthunder/FreeCAD_assembly3/wiki/Expression-and-Spreadsheet#spreadsheet-edit-mode). To get some feeling about it. Simply enter any quantity, e.g. 1mm, and right click the cell, enable `Quantity` edit mode. Now when you edit the cell, it will show a spin box, just like the property editor. To make the spin box always visible, you can enable `Persistent` edit option. `CheckBox` is another straightforward edit mode. Other modes may be a bit obscure, and requires some programming knowledge. You can checkout the demo code in the linked page.

herrgerd, 2021-04-05:
@realthunder thanks a lot for pointing to that page. Damn, I should stop relying solely on google to find information. Should've looked there!

MarcelP102, 2021-04-05:
@realthunder thanks for pointing me to the Wiki page. 

- Insert row between configuration table works indeed, I don't know what I did wrong before.
- Unsetup seems to be the way to overcome the entering row above the configuration table. I will use this method.
- I didn't know you could also rename the property group in the property view. Thanks for showing.

Found some more small things, if you like I can make a new issue? 
- Compared to official Freecad branch I found it weird that I couldn't type in the Content and Alias field at the top in your branch: 
[image]

- Also some themes/stylesheets show double text in the configuration table. (Behave-dark, Dark-contrast and Prodark)
[image]

- When typing inside a field your branch shows some style around the cell making the text invincible with some themes:
Like Prodark:  
[image] compared to official freecad: 
[image]
I understand that this is possible due the option you give to scroll within the cell. 

Or it's very hard to control the scroll buttons like in the Behave-dark theme: 
[image]

In the Dark-.. and Light-.. themes everything works fine.

LongLiveCHIEF, 2021-12-20:
> Found some more small things, if you like I can make a new issue?

Compared to the official Freecad branch I found it weird that I couldn't type in the Content and Alias field at the top in your branch:

I'm seeing the same thing in the release:

```
OS: Ubuntu 20.10 (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +3962 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: c6cb39697eb439caf7e5bd0208ca3082a7820547
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

```