realthunder/FreeCAD_assembly3 #557: Unable to delete quotes in sketch

Anducalliew, 2021-04-04:
With latest daily release(FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210328-glibc2.12-x86_64.AppImage) I'm not able to delete a quote in a sketch.

Open new file
Create a sketch
Draw a single line
Add a horizontal quote

I can't delete the created quote with delete, nor with dx click, delete.
If I save the file and open it with standard freecad 0.19 I can delete the quote.

kevenwyld, 2021-04-04:
By quote do you mean constraint?  Can you delete any constraints in a sketch?  I can't seem to delete anything except lines/points in the latest daily.  Even by selecting it in the constraint box and right click -> deleting.  It works as expected in previous versions.

Anducalliew, 2021-04-05:
Yes, I mean constraints. I can't delete any type of constraint.

realthunder, 2021-04-05:
This is duplicate of realthunder/FreeCAD#123. And it has been fixed already. I'll make a new release soon.

Anducalliew, 2021-04-05:
Thanks and sorry for the duplicate!

closed 2021-04-05 by Anducalliew