realthunder/FreeCAD_assembly3 #559: Impossible to delete constraints in sketchs V 28.03.2021

mwlandolsi, 2021-04-06:
Just open any sketch in Partdesign WB or Sketch WB.
constraints are displayed but not able to delete.
This issue is just at this update. for the last versions there is no problem.

OS: Linux Mint 19.3 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.22506 +5276 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ff1590e1d7c0aeaf45a29941ec20a10d54cf92ff
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

papamat, 2021-04-06:
I can also confirm this behavior, but I think that the problem is that the constraint can't be selected. You can double click on it, you can grab it, but if you just single click is getting selected and automatically dis-selected. You cannot delete it nor make it reference by pressing the button. 

OS: Fedora 33 (Workstation Edition) (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.22506 +5276 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ff1590e1d7c0aeaf45a29941ec20a10d54cf92ff
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

AcumenDev, 2021-04-11:
Hello, author already fixed this bug. Need to  wait new version build 
https://github.com/realthunder/FreeCAD/issues/123

closed 2021-05-05 by mwlandolsi