realthunder/FreeCAD_assembly3 #561: Value of angle constraint jumps to other constraint

oerkel47, 2021-04-06:
Hi.

I came across a bug that results in an angle from a constraint somehow getting applied to a neighbouring constraint (reinterpreted as mm) which can break the sketch. I could retrigger the bug after undoing it. It's related to the history of what you did before. 
The bug is kind of reproducable by just fiddling around with constraints related to the angle constraint but I did not manage to find out which events exactly lead to it. 

https://user-images.githubusercontent.com/57231822/113753810-d5f6f500-970e-11eb-949a-06840a72f050.mp4

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.24301 +3441 (Git)
Build type: Release
Branch: LinkStage3
Hash: 0a157a307c91d241a370cdda6d8f0471b7c48915
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a

OCC version: 7.5.0
Locale: German/Germany (de_DE)

luzpaz, 2021-07-17:
@oerkel47 can you still reproduce ? + could you provide a sample file?

oerkel47, 2021-07-17:
Hi. I played around with the problematic sketch and the current stable version but could not reproduce the problem. Is there some kind of release archive where I can download the old version to double check?

Otherwise I can not say 100%. I figure there is a good chance for it being fixed along the way.

Rabbit-sk, 2021-07-17:
> Is there some kind of release archive where I can download the old version to double check?

I can send you 7zip file of this release. You can download it [here](https://ulozto.net/tam/3d523194-3af4-4017-ac17-2e7caacc23b7).

Please let me know when you download the file, and I will then delete it from the data sharing service.

Rabbit-sk, 2021-07-17:
> Is there some kind of release archive where I can download the old version to double check?

I can send you 7zip file of this release. You can download it [here](https://ulozto.net/file/jmYEC7jcqIop/freecad-asm3-win64-py3-qt5-12-20210328-7z#!ZGR2BQR2AJZ4BQD1AQN4MwSwMGuxAUOiFKR3FTu3nx9SITR1Aj==7).

Please let me know when you download the file, and I will then delete it from the data sharing service.

oerkel47, 2021-07-17:
@Rabbit-sk Unfortunately that hoster is blocked here in Germany. Could you upload it somewhere else?

@luzpaz if you want to test yourself, this is the file I used for the video in april.
[BEARING_BUG.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6834701/BEARING_BUG.zip)

Rabbit-sk, 2021-07-17:
> Could you upload it somewhere else?

 https://www.uschovna.cz/en/zasilka/NPMV87IAUT2IWBW9-E22/

Rabbit-sk, 2021-07-21:
@oerkel47 So how's it going with double check in version 2021-03-28?

oerkel47, 2021-07-21:
@Rabbit-sk Unfortunately couldn't reproduce it even with the old version. So I really can't give you any meaningful conclusion. I can give it another try, but I don't expect quick results.