realthunder/FreeCAD_assembly3 #562: Broken model

gandy92, 2021-04-06:
A model with several groups and parts that I can open without errors in version 2021.0224, is completely broken (no tree structure, almost all sketches and coordinate systems are on the topmost level) when opened in 2021.0311 or 2021.0328. In the latter two versions a popup window informs me of errors when loading the model and possible data loss if the file is saved in this state. The only entry in the log is:
```
21:36:41  No profile linked
21:36:41  Linked shape object is empty
21:36:41  Invalid Document.xml: Linked shape object is empty
```
Is there any way to save this model? Would it help debugging to attach it here? What additional information should I provide?
Any help is highly appreciated.

Best,
Andy.

realthunder, 2021-04-07:
Yes, please send me the file to my email realthunder.dev at gmail.

gandy92, 2021-04-08:
@realthunder did you get my email? Just want to make sure I got the email address right.

realthunder, 2021-04-08:
Yeah, I got the email from junk folder. Will get back to you soon.

gandy92, 2021-04-20:
thank you, I just tried to load the model with v2021.0412 and it loads just fine. Shall I close the ticket?

closed 2021-04-22 by gandy92